Sleeper85/esphome-jk-bms-can #11: M5atom-lite

gutowscr471, 2024-01-27:
Had issues with the latest configuration files for wired.  Think you need to update to type: arduino

config/config_atom-lite-esp32-pico.yaml

 esp32:
  board: m5stack-atom
  framework:
    type: arduino

Sleeper85, 2024-01-28:
Noted. I will ask the person responsible for this platform to do some testing.

Sleeper85, 2024-01-28:
According to my research Atom Lite works well with the **esp-idf** framework.

What error do you receive exactly when compiling the YAML?

Sleeper85, 2024-02-04:
It actually seems that it is difficult to compile the YAML with Atom Lite (esp32-pico) and **esp-idf**.
Additionally Atom Lite is not stable with the BLE version. I changed to **arduino**.

Thanks

closed 2024-02-04 by Sleeper85