Sleeper85/esphome-jk-bms-can #12: some improvements

octane321, 2024-02-02:
works great so far - i have some improvements requests

- Use BMS max charge/discharge current (-5%) [tested and working]
- recover globals (charge/float voltage, absorbtion time etc) after restart (or is there any disadvantage like wearing?) [tested and working]
- add ability to change CAN Protokoll and Name dynamic (for testing inverter communication)
- use celltemperatures in a relevant way (if temp <20°C, use min temp of cells, if >20 and <40°C use average temp, if >40°C use max temp)
- seperate charge logic into new .yaml and include it as a package

what are your thoughts about that?

THX

Sleeper85, 2024-02-04:
Hi,

There are some interesting points like that of temperature, I'm about to add an "Alarms Logic" section see the links below.

I have a lot of requests for improvement and some will not suit everyone.
I think I should collect all the requests and hold a vote for each question.

I note your requests.

virus100b, 2024-02-04:
Hello,

@ Sleeper85 Congratulation for the nice project you have done 
I have moddifed liitle your code adding the option of modifying discharge current from slider. I am also planning to implement limiting discharging current to 10% from normal discharging current when battery cappacity drop below 15%. I know some BMS has this function. 

What is you oppinion about this?

Sleeper85, 2024-02-04:
I have already added an "Alarms Logic" section to address the [@arzaman](https://diysolarforum.com/members/69458/) and [@hxx](https://diysolarforum.com/members/21118/) balancing problem. I'm going to add temperature control to it. Thus, this section will aim to detect problems before the BMS cut-off which will prevent it from working its MOSFETs to cut a lot of amps.

Here's what will be part of version 1.16.5 :

- High temperature control (45°C) => stop charging and discharging
- Low temperature control (0°C) => stop charging
- Cell max voltage control (Bulk V. / 16) => reduce the charging amps to 5A and start a balancing phase
- Cell min voltage control (min_discharge_v /16) ? or bms_cell_undervoltage_protection + 0.x ? => stop discharging

This could be based on information from the BMS or values in the script.
If the BMS is poorly programmed then it will not be useful.

I can take advantage of this to add some of your requests such as the temperature sent (min or max) or a new slider for the discharge current.

But some of your requests are special and may not suit everyone.

FYI: I'm preparing for a long 4 month trip, version 1.16.5 will be the last one I can test before leaving.

octane321, 2024-02-04:
Didnt read all about the problems from arzaman and hxx but coult imagine what problems they have and the solution sounds good.
I would appreciate if the new Alarm logic will be based on the bms values - a poorly configured BMS is always a problem, but if you want to implement some values to set it would be not a bad idea.

i also understand if some requests will not suite to everyone - i´ll try to implement them for personal needs then if i´m able to.
Looking forward to 1.16.5 ;-)

I wish you a wonderful time on the trip

virus100b, 2024-02-04:
High temperature control (45°C) => stop charging and discharging, i believe is temperature of the cells measured by the two external probes, as the mosfets are not "suffering" by working at 45°C.
Regarding my plan with 10% of the current  limit function i can understand that it my not be suitable for others and i can tweak my code for that. For now i will wait for your final reales as the new feature are nice to have.

Some question:
1. Why did you name the MOS temperature "tube temperature"? Was a bit confusing at the begining as inside BMS they are called MOS.
2.   Discharging low temperature and Discharging low temperature recovery i cannot find in my BMS settings thus are present in the ESP home. This temperature may be hidden?

[image]

P.S. The inverter i have is a  Deye sun 12k-sg04lp3-eu. I have seen that is already in the confirmed working inverters, but i can reconfirms that is working fine ESP32 and TJA1050. 

For power the ESP i have use 12V that inverter provide on CN2 pins 7, 8 and a stepdown converter. Maybe for Deye owners my be useful info spare using a 48-5V Power supply.

octane321, 2024-02-04:
@virus100b Good point on the High temperature control
the 45°C should be triggered by the cell votages (external probes)
But there is also a MOS OTP and OTPR which could also trigger an alarm.
Would be great to limit the current by x% in both cases instead of shutting (dis)charging completely down

But i don´t get the point of limiting the current if SoC<15%
This will only push the inverter in a range where the efficiency is really bad due to the constant selfcomsumption. And it does not harm the cells or BMS to discharge at the normal rate - so why would you do that?

> 1. Why did you name the MOS temperature "tube temperature"? Was a bit confusing at the begining as inside BMS they are called MOS.

The transistors which control the charging and discharging function of the BMS are commonly called "MOS Tubes"

> 2\. Discharging low temperature and Discharging low temperature recovery i cannot find in my BMS settings thus are present in the ESP home. This temperature may be hidden?

these settings are only available on the JK Inverter BMS i think - but i might be wrong...there are so many different Hardware and Software Versions out there

Sleeper85, 2024-02-06:
@octane321 The temperature sent with ID 0x356 has an informative role. There are other parts in the code where the minimum or maximum temperature is communicated or used. So I opted to simply send the average temperature.

@virus100b I added the "Discharging current max" slider.

Compiled at home but haven't had time to do all the tests yet.

V1.16.5 Sleeper85 : Add Preventive Alarms Logic with Balancing function, CAN ID 0x356: send average temperature of T1/T2, new "Discharging current max" slider

```YAML
# +--------------------------------------+
# | Preventive Alarms Logic              |
# +--------------------------------------+
# Balancing max current, the charging current will be reduced to this value as soon as the voltage of a cell reaches 3.45v (bulk_voltage / 16)
  balancing_a: "5"
# +--------------------------------------+
# Information from the EVE LF280K v2 cell datasheet
# LFP Recommended Operating Temperature Range : 10~45°C
# Number of cycles at 25°C : 6000
# Number of cycles at 45°C : 2500
# 60°C  : Never charging or discharging
# 0°C   : Never charging
# -30°C : Never discharging
# +--------------------------------------+
# High Temp Protection - Temp > OTP : Stop Charging and Discharging
  otp: "45.0"
# Low Temp Protection  - Temp < UTP : Stop Charging
  utp: "1.0"
```

[JK-BMS-CAN 1.16.5](https://github.com/Sleeper85/esphome-jk-bms-can/files/14173400/jk-bms-can_1.16.5.zip)

virus100b, 2024-02-06:
Hello, 

@Sleeper85 Thanks for new version, if i have time i will try it today. Enjoy and good luck in your trip!

@octane321  "But i don´t get the point of limiting the current if SoC<15%.

I think the idea for limiting the current close to end of the capacity is because on high loads the cell voltage drops and you may switch off the discarding before reaching desired discharge capacity. 

On the charging the current is limiting itself because of constant voltage charging and  i see no point limiting unless poorly setup BMS command the inverter a higher voltage (or a fault in inverter) and then the current is obviously high.

Sleeper85, 2024-02-08:
Here is the latest version (beta) with the changelog.

If you have time to test it, I would like to have your feedback before publishing it on GitHub.

Thanks

[V1.16.6](https://diysolarforum.com/threads/jk-bms-can-bus-comms-now-possible-for-inverters-that-support-goodwe-and-pylontech-batteries.48963/page-31#post-1002153)

virus100b, 2024-02-09:
I see that balancing current entity is missing from the wired version. BMS is not exposing that parameter on 485 port?

Sleeper85, 2024-02-09:
Exactly, there are differences between the information available from BLE
and Wire.

virus100b, 2024-02-12:
Has anyone tested the code with ESP32 C3 board? I have readed that on S2 CAN is not working. How about C3?

shvmm, 2024-02-12:
> Has anyone tested the code with ESP32 C3 board? I have readed that on S2 CAN is not working. How about C3?

S2 will not work. The ESPRESSIF docs explicitly state that it lacks support for TWAI (CAN).

https://www.espressif.com/en/products/socs

Sleeper85, 2024-02-12:
Atom ESP32-S3 work.

octane321, 2024-02-16:
> Here is the latest version (beta) with the changelog.
>
> If you have time to test it, I would like to have your feedback before publishing it on GitHub.
>
> Thanks [...]

Hi @Sleeper85 ,
hope you are doing well!

Just spent the last couple of days to build an add-on with addressable lights to show the status and SoC - but more on that if its completely done.
I just wanted to test it with the new 1.16.6 beta but saw you removed the link on DIY Forum :-(

Have you already left? I wish you a great time on your trip

Sleeper85, 2024-02-17:
Hi,
I'm leaving next week.
The latest version 1.17.2 will be available soon. And if you want I can share it with you privately if you want to test it ;-)

virus100b, 2024-02-17:
Hi @Sleeper85 ,

Happy to test, even i am 2000km away of my home i have remote access of my HA so i can flash it.
What are the goodies for the version 1.17.2 ?

In the future i want to build another battery pack  as i will  like more storage capacity  (currently i have a 280Ah battery) and i will like to monitor that battery too. 
I will try to modify the code for that at some point. 
I believe the simplest way is use the same board using the one free UART for connecting to the second BMS and send to inverter the sum of SOC, the rest of the settings will be the same. HA entities can be published separately(e.g. SK#01... and JK#02...) Right?

Sleeper85, 2024-02-17:
We still need to fix one more thing before releasing the latest version. It will be soon.

@syssi offers multi-devices UART and BLE YAML. I monitor 2 JK-BMS (BLE) with a single board (without CAN) so yes it's possible. Concerning UART, it's maximum 3 connections with ESP32.

- V1.17.2 MrPablo   : Added function "Auto Charge/Discharge Current Control" to avoid OVP/UVP alarms
- V1.17.1 Sleeper85 : New Cut-Off Current/Voltage Charging Logic for LFP with the participation of @shvmm
- V1.16.6 Sleeper85 : Selectable CAN settings + Adding inverter_offset_v + Improved CAN ID 0x355, sending 100% only at the end of the absorption phase, adding bytes [04:05] and [06:07] + Automatic calculation of the number of battery modules + Save and Restore slider values
- V1.16.5 Sleeper85 : Add Preventive Alarms Logic, CAN ID 0x356: send average temperature of T1/T2, new "Discharging current max" and "Balancing current max" slider

virus100b, 2024-02-17:
@Sleeper85 

> 1.  I monitor 2 JK-BMS (BLE) with a single board (without CAN) so yes it's possible. Concerning UART, it's maximum 3 connections with ESP32.  

This option is good but i want to send sum of  both batteries SOC to the Inverter as i using Time of use table to stop discharging the battery.

I have modified the name for cells 1-to 9 in 01 to 09 as they where not showing in the right order on entities list. 1, 10...so on, you may modify that if you like in the last update.
[image]

octane321, 2024-02-17:
> Hi, I'm leaving next week. The latest version 1.17.2 will be available soon. And if you want I can share it with you privately if you want to test it ;-)

Good morning, would be very nice to get the actual version - so i can implement my things.
I don´t know if you can send it on github somehow - but i´m also on diy forum (sent you a PM)

octane321, 2024-02-17:
> @Sleeper85
> 
> > 1. I monitor 2 JK-BMS (BLE) with a single board (without CAN) so yes it's possible. Concerning UART, it's maximum 3 connections with ESP32.
> 
> This option is good but i want to send sum of both batteries SOC to the Inverter as i am using "Time of use" table to stop discharging the battery.
> 
> I have modified the name for cells 1-to 9 in 01 to 09 as they where not showing in the right order on entities list. 1, 10...so on, you may modify that if you like in the last update. ![cells](https://private-user-images.githubusercontent.com/64686560/305629399-af43e91b-c328-48d1-9611-e25fccd7166a.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgxNjQ0MzQsIm5iZiI6MTcwODE2NDEzNCwicGF0aCI6Ii82NDY4NjU2MC8zMDU2MjkzOTktYWY0M2U5MWItYzMyOC00OGQxLTk2MTEtZTI1ZmNjZDcxNjZhLkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE3VDEwMDIxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBiOTQ5MjJhMDQ2NzM1Zjk1OTY3YWMxOTM3ZDkyMWZjNTVjNDE0MTgzNTU3OTliNDhjZjEzYjQ0NDA5MmZiOTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.sho03JW7h992mmWqv7AT7eAWp83MGlIQSAhtFyZpUO8)

i renamed them too for the correct order (the little Mr.Monk in me strictly demanded that)

I have 3 BMS (with 280Ah cells) and i tried to get two BMS running on a normal ESP32 with BLE, but the BLE Overhead is just massive so quickly running out of heap when using a addressable light strip too.
I deceided to switch to an ESP32-S3 N16R8 now with 8MB of PSRAM. Even if the PSRAM is completely empty, i got much more free heap now. Maybe i´ll give it another try connecting all 3 BMSs together with the CAN communication.
atm i´m running 4 devices - but it would be much safer if only one device deals with everything.

Sleeper85, 2024-02-17:
@virus100b @octane321 

Ok, cell number renaming will be part of the next release.

The new YAMLs will be published early next week.

Sleeper85, 2024-02-23:
[Version 1.17.3 is released](https://github.com/Sleeper85/esphome-jk-bms-can/releases/tag/V1.17.3)