Sleeper85/esphome-jk-bms-can #13: round requested bulk and float voltages

octane321, 2024-02-02:
is there a way to get rid of the decimals on values from Float and Bulk voltage?
Values should be 56.2V and 55.2V in this example

[image]

I could not find a filter or stock setting to fix this - any idea?
(This is just a cosmetic issue)

Sleeper85, 2024-02-04:
Hi,

From what I understand, these are values that were modified by the slider?
How many decimals do you see just after starting the ESP32 without changing them with the slider?

octane321, 2024-02-04:
the values are right after a restart of the ESP (untouched) - you can also see these values in the webUI if you switch from slider to box

luckylinux, 2024-03-26:
> the values are right after a restart of the ESP (untouched) - you can also see these values in the webUI if you switch from slider to box

I can observe the same issue ... Float/Bulk Voltage slider has like 5 decimals.

For a "Sensor" type component you either could use `round` or `accuracy_decimals`:
https://esphome.io/components/sensor/index.html

For Number it's more complicated. I'm definitively not an ESPHome expert, but what about using a lamba and round(1) ?
https://esphome.io/components/number/#number-on-value

Maybe like in one of the posts here ?
https://community.home-assistant.io/t/solved-use-filter-with-maximum-value/426872

shvmm, 2024-04-05:
Highly likely a well known REST API issue.

Relavant issue on esphome repo.
https://github.com/esphome/esphome-webserver/pull/57

This issue can be safely closed.

luckylinux, 2024-04-05:
Not sure if the esphome-webserver fix solves the issues in Home Assistant as well though.

@shvmm Do you have more details ? 

And when will this be "available" when building the `esphome-jk-bms-can` project ? I'm currently running ESPHome 2023.12.5 so it's definitively lagging behind. Is the ESPHome version (installed via `pip`) lagging so much behind ? Or a specific version of ESPHome is "pinned" either by `esphome-jk-bms-can` or syssi `esphome-jk-bms` ?

shvmm, 2024-04-05:
As far as I am aware, home assistant isn't affected by this issue.

luckylinux, 2024-04-05:
> As far as I am aware, home assistant isn't affected by this issue.

Oh yeah, I am seing 5-6 decimals for float / bulk voltages etc.

Sleeper85, 2024-07-18:
Can I close this issue ?

luckylinux, 2024-07-18:
@Sleeper85 I still see 5 decimals in the "sliding Bar" (Settings e.g. for Bulk/Float Voltage), so I would say the Issue is not solved. You can argue it's not the "fault" of your Code though, it's probably some upstream (Home Assistant and/or ESPHome) Issue.

Sleeper85, 2024-07-18:
ESPhome 2024.7.0 and YamBMS 1.4.1

[image]

luckylinux, 2024-07-18:
That's not Home Assistant, it's the Web Server (which I disabled to avoid further Hangups on the ESP32).

[image]

Home Assistant 2024.6.1 and ESPHome 2024.5.0 if that Helps ...
[image]

Sleeper85, 2024-07-18:
> ESPhome 2024.7.0 and YamBMS 1.4.1 [...]

@luckylinux

It's a Web server screenshot.

closed 2024-07-24 by Sleeper85