Sleeper85/esphome-jk-bms-can #27: Atom S3 Lite - configuration inconsistency

marekorok, 2024-03-21:
Hi there,
I try to make it work on recommended Atom S3 Lite with CAN module but no luck, then I found some inconsistency in the config files which don't even compile, for example there is a led defined on pin 2 while GPIO02 is ocupied by UART-TTL rx_pin, second S3 Lite does not have led on pin 2, instead it has RGB led ws2812 on gpio35.

Then trying BLE version without uart and removing led the code compile but does not run (boot loop? wdt?), i hear usb device connecting and disconnecting on the pc constantly, I can force boot mode with reset button and upload new code but that's it, no luck so far

checked the dev branch, nice progress but same issues

What am I missing here?

marekorok, 2024-03-22:
ok, I made some progress here

seems ESP32 S3 and esphome has generic problems with platform esp-idf (with arduino it works), adding build options to esphome sections helps:
```
esphome:
  platformio_options:
    board_build.flash_mode: dio
```

now the code works, no wdt anymore

for the led I found this example code and trying to implement it:
https://gist.github.com/tathamoddie/62afc41e5b4464f5adeef4b580fb3294

cheers

arzaman, 2024-03-23:
> ok, I made some progress here
> 
> seems ESP32 S3 and esphome have generic problems with platform esp-idf (with arduino it works), adding build options to esphome sections helps:
> 
> ```
> esphome:
>   platformio_options:
>     board_build.flash_mode: dio
> ```
> 
> now the code works, no wdt anymore
> 
> for the led I found this example code, tested successfuly standalone and now trying to implement it: https://github.com/wildekek/ESPHome-M5Stack-Atom-S3-Lite/blob/main/M5Stack-AtomS3-Lite.yaml
> 
> cheers

Hi I was able to compile succesfully the Yaml version 1.17.4  using both arduino framework and espressif one
for the espressif ide I have used this declaration

`esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
  variant: esp32s3`

I have installed the toolchain on windows enviroment and I compile the file via command line

Davide

marekorok, 2024-03-23:
> Hi I was able to compile succesfully the Yaml version 1.17.4 using both arduino framework and espressif one for the espressif ide I have used this declaration
> 

it's not a problem with compilation but once you upload the code to the device (in this case Atom S3 lite, Esp32S3) it enters boot loop

more info here:
https://github.com/esphome/issues/issues/4850

I have ignored the led for the moment, the remaining code works fine now in bluetooth mode

If you want to gpio conflict issue test then try to compile:
[ESP32_LFP_Wire_jk-bms-can.yaml](https://github.com/Sleeper85/esphome-jk-bms-can/blob/main/ESP32_LFP_Wire_jk-bms-can.yaml)

and apply changes recommended for Atom S3 Lite:
https://github.com/Sleeper85/esphome-jk-bms-can/blob/main/config/config_atom-s3-lite-esp32-s3.yaml

```
# +--------------------------------------+
# | ESP32 CAN/serial port pins           |
# +--------------------------------------+
# GPIO pins your CAN bus transceiver ATOMIC CANBus Base (CA-IS3050G)
  can_tx_pin: GPIO5
  can_rx_pin: GPIO6
# GPIO pins your JK-BMS UART-TTL is connected to the grove port of Atom Lite
  tx_pin: GPIO1
  rx_pin: GPIO2 <------------ here is conflict gpio
```

cheers

cheers

arzaman, 2024-03-23:
> 

>   rx_pin: GPIO2 <------------ here is conflict gpio
> ```
> 
> cheers

For the wired version I have changed the led GPIO in this way to avoid conflict
notice that Atom S3 led is a RGB led and is not manged at the moment

`output:
  - platform: gpio
    pin: 35
    id: led
    inverted: true`

marekorok, 2024-03-26:
> @arzaman
> 
> Concerning the platform, do you confirm that it works well with esp-idf ? You had done all the tests and it seemed OK ?

No issues so far, yet I tested only bluetooth connection to the jk bms BD6A20S10P and it is pretty stable, no single glitch after few days.

Waiting for my inverter to verify CAN and later rs485, but wondering if need it while bluetooth works ok

In general very positive experience, thank you!

cheers

closed 2024-03-26 by marekorok