Sleeper85/esphome-jk-bms-can #34: Current control oscillation

luckylinux, 2024-03-26:
[image]

[image]

[image]

OVPR is set at 3.52 VDC:
[image]

Detail of max cell voltage
[image]

luckylinux, 2024-03-26:
Sure, I removed the overvoltage protection from the plots that are for single cells (much lower of course)

[image]

MrPabloUK, 2024-03-26:
So at the time of the current oscillation, your float voltage was 56.1v, or 3.506v per cell?

luckylinux, 2024-03-26:
> So at the time of the current oscillation, your float voltage was 56.1v, or 3.506v per cell?

Yeah, I needed to raise it because otherwise it will never balance out ...

MrPabloUK, 2024-03-26:
With the following settings:
- Float = 3.506v
- OVPR = 3.520v
- End of Curve Factor = 2.0
- Curve Shape = 0.4
- Max Charge Current = 180A

You end up with a curve like this:
[image]

Those settings result in a request to reduce current 180A over 14mV. That's not realistic, even if it were linear then it'd be nearly 13A per mV.

You should set float back to 3.37v or so, expanding the range over which the current will be reduced.
I know you've had issues keeping the charge state in bulk or balancing - my suggestion is to lower the balance trigger voltage to below 3.45v as previously stated.
The changes mentioned in https://github.com/Sleeper85/esphome-jk-bms-can/issues/35 may also help.

luckylinux, 2024-03-26:
But if I recalled correctly, previously (yesterday ?), this was also happening during Bulk, not only Float ...

MrPabloUK, 2024-03-26:
The code for this feature uses the cell float voltage as the start point of current tapering.
This is why I suggest setting your float voltage back to a more reasonable setting. Artificially increasing the float voltage in order to facilitate balancing shouldn't be necessary, assuming the balance trigger voltage is appropriately set.

luckylinux, 2024-03-26:
I agree. Problem is that even though I properly balanced it 3 weeks ago, it unbalanced quite quickly since then (and I don't think I went above the required knee point of the charge characteristic, when there was no communication between Inverter and BMS).

MrPabloUK, 2024-03-26:
For now, I'll close this issue out as the code itself is not causing the problem.
Thanks for testing out an edge case!

closed 2024-03-26 by MrPabloUK