Sleeper85/esphome-jk-bms-can #35: Premature triggering of EOC

luckylinux, 2024-03-26:
Please note that Balancing is ON yet the ESP32 went into FLOAT charging status now.

[image]

Probably this happens because for a second or so (probably clouds caused the battery to discharge for a few seconds) the cell voltage went below balance trigger voltage, due to the battery being discharged (either because of clouds or load step):
[image]

[image]

MrPabloUK, 2024-03-26:
@luckylinux, I have been testing a cycle / time threshold for triggering EOC, you can see the change to the code at https://github.com/Sleeper85/esphome-jk-bms-can/commit/d4f257c0c6f2087cd41c3afcfed662160467edf6

Please note that this is currently a 60 cycle threshold and this section of code is generally run once per second.
This will not prevent EOC from being triggered if current reduction is due to generation or load changes if they last for more than this time.

MrPabloUK, 2024-03-26:
Please note that this logic is liable to change in the future packaged version under development.

MrPabloUK, 2024-03-26:
You can see it in https://github.com/Sleeper85/esphome-jk-bms-can/compare/35-premature-triggering-of-eoc, applied to a single YAML for now.

If you can test the code first with the default 60 cycles, that would be appreciated.
If required, you can then manually adjust this value - future code versions will likely have it as a parameter for user adjustment.

I would caution against a period as long as 30 minutes however, that will likely overcharge your cells as per the detail in https://nordkyndesign.com/charging-marine-lithium-battery-banks/.

luckylinux, 2024-03-26:
OK so basically backport the "delta" into my version, got it.

What do you mean by "overcharge your cells" ? Isn't that how top balance works anyways ? And as long as all cells < 3.65 VDC it shouldn't overcharge ...

luckylinux, 2024-03-26:
@MrPabloUK: What would happen if now, I disable BLE connection from ESP32 to JK BMS (concerning CAN Communication) ? Would it cause the inverter to trip (BMS_Err_Stop is SET now).

MrPabloUK, 2024-03-26:
Honestly, I don't know for certain. I use a wired connection as I don't quite trust BLE for this project.
The change made in the previously mentioned branch may be released in the next sub-release, once it's been tidied up.

luckylinux, 2024-03-26:
Question is ... If I were to implement multi BMS like Der_Hannes did, should I use this as a base or the development branch ?

MrPabloUK, 2024-03-27:
Closing issue as feature has been added to V1.17.5

closed 2024-03-27 by MrPabloUK