Sleeper85/esphome-jk-bms-can #38: Potential naming issues for sensors - Duplicate device name / hostname

luckylinux, 2024-03-27:
I previously mentioned this issue regarding Battery 02, when upgrading from Syssi esphome-jk-bms to Sleeper85 esphome-jk-bms-can.

I now flashed also Battery 01 in order to have more consistent readings across the different batteries.

The essential problem is that the hostname is repeated in the sensor.xxx access string.

But it's not always the case. Some are, some are not.

I deleted the previous MQTT Device (ESP32 board) after reflashing this one (Atom S3 Lite Board).

[image]

```
type: vertical-stack
cards:
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat01_jk_bms_bat01_power
      - entity: sensor.jk_bms_bat02_power
    title: Power
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat02_jk_bms_bat02_requested_charge_current
      - entity: sensor.jk_bms_bat02_jk_bms_bat02_requested_charge_voltage
      - entity: sensor.jk_bms_bat02_jk_bms_bat02_requested_discharge_current
    title: Requested Charge Parameters
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat01_jk_bms_bat01_current
      - entity: sensor.jk_bms_bat02_current
    title: Current
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat01_state_of_charge
      - entity: sensor.jk_bms_bat02_state_of_charge
    title: State of Charge
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat01_max_cell_voltage
      - entity: sensor.jk_bms_bat02_max_cell_voltage
    title: Maximum Cell Voltage
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat01_min_cell_voltage
      - entity: sensor.jk_bms_bat02_min_cell_voltage
    title: Minimum Cell Voltage
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat01_delta_cell_voltage
      - entity: sensor.jk_bms_bat02_delta_cell_voltage
    title: Cell Delta Voltage
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat01_temperature_sensor_1
      - entity: sensor.jk_bms_bat01_temperature_sensor_2
      - entity: sensor.jk_bms_bat02_temperature_sensor_1
      - entity: sensor.jk_bms_bat02_temperature_sensor_2
    title: Ambient Temperature

```

YAML file used with Syssi esphome-jk-bms: https://github.com/luckylinux/jk-bms-build-helpers/blob/main/_old/esphome-jk-bms/esp32-ble-2.0.0.yaml
YAML file used with Sleeper85 esphome-jk-bms-can: https://github.com/luckylinux/jk-bms-build-helpers/blob/main/esphome-jk-bms-can/esp32-ble-1.17.5.yaml

[image]
[image]

For reference (since I didn't have screenshots from before apart from those on the DIYSolarForum - will try to include those if I can find them), my weather Station Sainlogic WS3500 via Weewx and using weewx-mqtt reports like this to the Mosquitto MQTT Server:
[image]

luckylinux, 2024-03-27:
I couldn't find screenshots for Battery 02 having that issue before ...

Maybe related to https://community.home-assistant.io/t/esphome-mqtt-devices-have-duplicated-names/609858/4 ?

luckylinux, 2024-03-27:
What's worse is that it's not consistent and even in MQTT it shows one thing, in home assistant another.

I updated the simplified view for now this works, but it's definitively a hack:

[image]

```
type: vertical-stack
cards:
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat01_jk_bms_bat01_power
      - entity: sensor.jk_bms_bat02_power
    title: Power
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat02_jk_bms_bat02_requested_charge_current
      - entity: sensor.jk_bms_bat02_jk_bms_bat02_requested_charge_voltage
      - entity: sensor.jk_bms_bat02_jk_bms_bat02_requested_discharge_current
    title: Requested Charge Parameters
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat01_jk_bms_bat01_current
      - entity: sensor.jk_bms_bat02_current
    title: Current
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat01_jk_bms_bat01_state_of_charge
      - entity: sensor.jk_bms_bat02_state_of_charge
    title: State of Charge
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat01_jk_bms_bat01_max_cell_voltage
      - entity: sensor.jk_bms_bat02_max_cell_voltage
    title: Maximum Cell Voltage
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat01_jk_bms_bat01_min_cell_voltage
      - entity: sensor.jk_bms_bat02_min_cell_voltage
    title: Minimum Cell Voltage
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat01_jk_bms_bat01_delta_cell_voltage
      - entity: sensor.jk_bms_bat02_delta_cell_voltage
    title: Cell Delta Voltage
  - type: entities
    entities:
      - entity: sensor.jk_bms_bat01_jk_bms_bat01_temperature_sensor_1
      - entity: sensor.jk_bms_bat01_jk_bms_bat01_temperature_sensor_2
      - entity: sensor.jk_bms_bat02_temperature_sensor_1
      - entity: sensor.jk_bms_bat02_temperature_sensor_2
    title: Ambient Temperature
```

[image]
[image]

I tried:
- Restarted ESP32 of Battery01 (NEVER touch ESP32 of Battery02 / MASTER, otherwise grid goes down).
- Restarted Mosquitto MQTT Broker
- Restarted Home Assistant

That doesn't seem to help. There is something new going on here. Maybe a new library of ESPHome / IDF Framework is used compared to Syssi previous code ? Not sure.

luckylinux, 2024-03-27:
It's also even more funky because Battery 01 has some max/min/avg built in, Battery 02 does not ...

[image]

[image]

MrPabloUK, 2024-03-27:
Here's what I would try.
- Remove all JK BMS devices from HA
- Delete all discovery topics related to the JK BMS devices from your MQTT broker
- Restart HA
- Add auto detected devices in HA

luckylinux, 2024-03-27:
Concerning point 1: Remove all JK BMS devices from HA. Would they be re-added automatically ? I cannot afford to restart the ESP32 of battery 02 ...

Maybe I'll try your procedure with Battery 01 first.

luckylinux, 2024-03-27:
OK Deleted Battery 01 and the discovery topic on the MQTT Server for Battery 01
[image]
[image]

Restarted Home Assistant -> No joy
Restarted Mosquitto MQTT Broker -> Device got auto-detected

[image]

But still
[image]

Sleeper85, 2024-05-18:
Is this related to the issue below?

https://github.com/Sleeper85/esphome-jk-bms-can/issues/46

This has been corrected in the new version in testing phase.

All name and id are unique and prefixed with ${variable}.

Can I close this issue ?

luckylinux, 2024-05-18:
Not really.

From a quick Look I'd say that #46 is for a VERY SPECIFIC set of entities (e.g. `ignoring button.jk_bms_bat02_jk_bms_esp32_restart` from the Logs, probably because those entities did NOT have an ID "set" in the YAML File), whereas the Issue reported here is basically for ALL (or most of) Entities (which had an ID "set" in the YAML File, but it was not unique).

But I guess a bit too much time went by.

Sleeper85, 2024-05-18:
So, I close this issue.

If you still have a similar problem you can re-open this issue.

closed 2024-05-18 by Sleeper85