Sleeper85/esphome-jk-bms-can #46: Non-Unique Identifiers/Names and possibly missing .state when grabbing id(entity) ?

luckylinux, 2024-04-18:
I noticed the following Items in my HomeAssistant Logs:
```
2024-04-18 21:10:21.231 ERROR (MainThread) [homeassistant.components.button] Platform mqtt does not generate unique IDs. ID ESPbuttonjk-bms_esp32_restart already exists - ignoring button.jk_bms_bat02_jk_bms_esp32_restart
2024-04-18 21:10:21.472 ERROR (MainThread) [homeassistant.components.select] Platform mqtt does not generate unique IDs. ID ESPselectcan_bms_name already exists - ignoring select.jk_bms_bat02_can_bms_name
2024-04-18 21:10:21.472 ERROR (MainThread) [homeassistant.components.select] Platform mqtt does not generate unique IDs. ID ESPselectcan_protocol already exists - ignoring select.jk_bms_bat02_can_protocol
2024-04-18 21:10:22.515 WARNING (MainThread) [homeassistant.components.mqtt.sensor] Invalid state message '1490107.0' from 'jk-bms-bat02/sensor/jk-bms-bat02_last_complete_charge/state'
2024-04-18 21:10:22.522 WARNING (MainThread) [homeassistant.components.mqtt.sensor] Invalid state message '1098780.0' from 'jk-bms-bat01/sensor/jk-bms-bat01_last_complete_charge/state'
```

The ERROR is probably referring to missing ${name} in the `id: can_protocol` I think, which would make it unique when other BMSs are Present on the HomeAssistant API / MQTT Network:
```
  - platform: template
    name: CAN Protocol
    id: can_protocol
    options:
      - "GENERAL"
      - "PYLON 1.2"
      - "PYLON +"
      - "SMA"
      - "VICTRON"
    restore_value: true
    initial_option: "PYLON +"
    optimistic: true
    entity_category: config
    set_action:
      - logger.log:
          format: "Chosen option: %s"
          args: ["x.c_str()"]
```

While the WARNING is probably complaining about the following piece of Code:
```
  - platform: template
    name: "${name} Last Complete Charge"
    device_class: timestamp
    id: eoc_timestamp_template
    lambda: return id(eoc_timestamp);
    on_value:
      then:
        - text_sensor.template.publish:
            id: eoc_timestamp_human
            state: !lambda |-
              char str[19];
              strftime(str, sizeof(str), " %H:%M %d %b %Y", localtime(&id(eoc_timestamp)));
              return  {str};
```

I thought that usually when accessing values we have to use `id(entity).state`.

For some other stuff I do with ESPHome I enforce **every time** that ID and Names are Unique (by prefixing them with the `name` (for Name) respectively a version of `name` without spaces and special Characters (for ID).

I think as a general Rule this should be pretty straightforward to implement as well as a quick fix :+1: .

Sleeper85, 2024-04-18:
Hi,

The errors seem to be related to the MQTT version of Der_Hannes. 

All IDs in the latest version 1.17.5 are unique. And the IDs of the packaged version in development are even better.

Concerning the .state this only concerns sensors, not global variables.

luckylinux, 2024-04-18:
Hi,

This is the "legacy" section of code from the main Repo https://github.com/Sleeper85/esphome-jk-bms-can (main) that I have adapted with some additional Number User-Tuneables plus some Patches provided by @MrPabloUK.

The source code that I copy-pasted is the one from the current main Repository.

And it matches my current (now probably a few versions behind) code.

That's why I think the issue is still present in your current/main branch ...

Sleeper85, 2024-04-19:
So the problem occurs because you have two ESP32s with the same code and you are using MQTT instead of api?

luckylinux, 2024-04-19:
If I **exclude** the mqtt WARNINGS (which I also get on the GARAGE/Production HomeAssistant Version), this is what I get on the GARAGE/Production HomeAssistant Logs:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun home-assistant (no readiness notification)
s6-rc: info: service legacy-services successfully started
2024-04-11 10:09:50.564 ERROR (MainThread) [aiodhcpwatcher] Cannot watch for dhcp packets: [Errno 1] Operation not permitted
2024-04-11 10:09:53.379 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.deye_b_sun12k_internal_reactive_power_generation_of_today (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'kVarh' which is not a valid unit for the device class ('energy') it is using; expected one of ['GJ', 'MJ', 'Wh', 'kWh', 'MWh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
2024-04-11 10:09:53.458 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.deye_c_sun12k_internal_reactive_power_generation_of_today (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'kVarh' which is not a valid unit for the device class ('energy') it is using; expected one of ['GJ', 'MJ', 'Wh', 'kWh', 'MWh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
2024-04-11 10:09:54.518 ERROR (MainThread) [homeassistant.components.homeassistant_alerts] Error requesting homeassistant_alerts data: Cannot connect to host alerts.home-assistant.io:443 ssl:default [Try again]
2024-04-11 10:09:55.456 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.deye_a_sun12k_internal_reactive_power_generation_of_today (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'kVarh' which is not a valid unit for the device class ('energy') it is using; expected one of ['GJ', 'MJ', 'Wh', 'kWh', 'MWh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
2024-04-11 10:10:48.845 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for deye-meteo-test-a @ 172.27.20.1: Timeout while connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='172.27.20.1', port=6053))] (TimeoutAPIError)
2024-04-11 10:10:48.845 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for esp32-s3-base @ 172.27.30.1: Timeout while connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='172.27.30.1', port=6053))] (TimeoutAPIError)
2024-04-11 10:12:42.047 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 350, in data_received
    messages, upgraded, tail = self._request_parser.feed_data(data)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "aiohttp/_http_parser.pyx", line 557, in aiohttp._http_parser.HttpParser.feed_data
aiohttp.http_exceptions.BadStatusLine: 400, message:
  Invalid method encountered:

    b'\x16\x03\x01\x02z\x01'
      ^
2024-04-11 10:21:08.900 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for jk-bms-bat01 @ 172.22.10.1: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='172.22.10.1', port=6053))]: [Errno 111] Connect call failed ('172.22.10.1', 6053) (SocketAPIError)
2024-04-11 10:21:47.597 WARNING (MainThread) [aioesphomeapi.connection] deye-c @ 172.22.20.3: Connection error occurred: [Errno 104] Connection reset by peer
2024-04-11 10:21:47.969 WARNING (MainThread) [aioesphomeapi.connection] deye-a @ 172.22.20.1: Connection error occurred: [Errno 104] Connection reset by peer
2024-04-11 10:21:48.000 WARNING (MainThread) [aioesphomeapi.connection] deye-b @ 172.22.20.2: Connection error occurred: [Errno 104] Connection reset by peer
2024-04-11 10:23:47.651 WARNING (MainThread) [aioesphomeapi.connection] deye-c @ 172.22.20.3: Connection error occurred: [Errno 104] Connection reset by peer
2024-04-11 10:23:56.499 WARNING (MainThread) [aioesphomeapi.connection] deye-b @ 172.22.20.2: Connection error occurred: [Errno 104] Connection reset by peer
2024-04-11 10:23:58.035 WARNING (MainThread) [aioesphomeapi.connection] deye-a @ 172.22.20.1: Connection error occurred: [Errno 104] Connection reset by peer
2024-04-12 18:35:45.418 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
[... cut ...]
But maybe that isn't enough for the ESP32, which seems to require Manual Configuration anyways:
```
time:
  - platform: sntp
    id: sntp_time
    timezone: Europe/Sofia
    servers:
     - 0.pool.ntp.org
     - 1.pool.ntp.org
     - 2.pool.ntp.org
```

So maybe I need to set:
```
time:
  - platform: sntp
    id: sntp_time
    timezone: Europe/Sofia
    servers:
     - 172.22.1.1
```

Still a bit weird that the RAW Value shows `Unknown`. 
Is it working correctly in your case ? Can you provide a Screenshot please ?

Sleeper85, 2024-04-19:
What you point out is correct. Sensor names and ids can be enhanced to be unique. There are different ways to do it and I've already made a lot of changes in the new packaged version that I'm focusing on now.

Regarding the eoc_timestamp code it's a few things that was made by @MrPabloUK and I haven't tested it personally but I assume that @MrPabloUK tested it before publishing it.

luckylinux, 2024-04-19:
I agree, it's not the highest priority.

It's just flooding the logs of HomeAssistant. Which is an Issue if then we don't see more Important / Real Issues there.

Sleeper85, 2024-04-19:
I will be careful not to have this problem in the new packaged version.

Thank you for reporting this issue.

We also look at the sntp config.

luckylinux, 2024-04-19:
@MrPabloUK Alright ... So it's probably DHCP Option 42 not being supported by ESPHome I guess.

Actually it's probably me that has to learn something new ...

Look at:
https://www.cisco.com/c/en/us/td/docs/net_mgmt/prime/network_registrar/9-0/dhcp/guide/DHCP_Guide/DHCP_Guide_appendix_01101.pdf

https://blog.abysm.org/2020/06/human-readable-dhcp-options-for-dnsmasq/

[image]

vs

[image]

Maybe also
[image]

Although SNTP is IPv6 (option 31) vs NTP is IPv4 ...

Google may indicate that DHCP IPv4 Server does NOT support SNTP Options:
[image]

So overall SMTP is NOT (exactly) the same as NTP, even though both are used for Time Synchronization.

Sleeper85, 2024-05-17:
This has been corrected in the new version in testing phase.

All name and id are unique and prefixed with ${variable}.

closed 2024-05-17 by Sleeper85