Sleeper85/esphome-jk-bms-can #48: Possible Glitch / Timing issue

luckylinux, 2024-05-09:
The System has been running quite well since approximatively 3 weeks now.

However today, out of the blue, the Inverter Tripped due to BMS Communication Failure (BMS-Err_Stop enabled on Deye Inverter).

Looking at Home Assistant Dashboard the main thing that caught my eye is this:
[image]

`14:24:44` is the time where the Issue occurred.

Looking at the logbook there are plenty of Entries like this every day (yesterday was 22 times, so basically 11 couples of brief moments on/off for a few seconds).

Not sure why it happened.

Also weird is that the ESP32 (both of them, both the one of the "dumb" battery and the one connected to the Deye Inverter) seem to have rebooted, if we can trust the Diagnostic Data.

ESP32 not Connected to any Inverter:
[image]

ESP32 Connected to the Deye Inverter:
[image]

This should not happen, since both ESP32s and the Rock 5B SBC that runs Home Assistant, MQTT etc, are connected to a 230VAC UPS. The Rock 5B SBC didn't reboot:
```
18:10:11 up 20 days,  1:12,  1 user,  load average: 0.80, 1.01, 1.01
```

luckylinux, 2024-05-17:
I'm also struggling with the ESP32 just rebooting/crashing/freezing thus causing an Inverter Trip, whenever I reboot the WiFi.

At first I thought disabling the fallback ap and the `captive_portal` could have fixed this (according to https://github.com/esphome/issues/issues/1679 `reboot_timeout` is ignored in case where the fallback ap is enabled).

Today I rebooted the WiFi again after applying those fixes and ... ESP32 Crashed/Rebooted/Freezed again and of course the Inverter Tripped.

@Sleeper85 , @MrPabloUK: Is there some timing Issue connected to WiFi loss, whereby the code automatically triggers a Restart ?

Configuration File (finally SNTP Time Sync and Home Assistant Time Sync work correctly by the way): https://github.com/luckylinux/jk-bms-build-helpers/blob/main/esphome-jk-bms-can/esp32-ble-1.17.5.yaml

luckylinux, 2024-05-17:
This Commit seems to Fix the issue related to WiFi AP (and/or Home Assistant Server and/or MQTT Server and/or ... whatever):

https://github.com/luckylinux/jk-bms-build-helpers/commit/b60d922b7966af904dc10909634094d517847a9d

Other Remarks ...
OK After testing stopping each Service Individually, I took it one step further and stopped them ALL one by one. Still no Trip / Watchdog event.

Issued 5 Reboots (from a Normal Operating State) now and nothing happened again.

It's now Working [for now] as you (and I) would expect ...

NOT sure in the end what really helped:
- Setting MQTT keepalive: 5s instead of default (15s)
- Setting Watchdog Timeout 30s instead of default (1s ???)
- Setting WiFi: power_save_mode: light and fast_connect: true

I also had a configuration issue in /etc/dhcpcd.conf with a Typo on an Interface Name (NOT related to either WLAN or LAN).
Furthermore there was a configuration issue in my "Headless" Management Script `/usr/local/sbin/check-network.sh` which tried to bring up/down the wrong (non-existing) Interface Name to try to ping the LAN Gateway
```
#!/bin/bash

gateway="192.168.1.1"
interface="eth0"

ping -c4 $gateway > /dev/null

if [ $? != 0 ]
then
  echo "No network connection, restarting $interface"
  /sbin/ifdown "$interface"
  sleep 5
  /sbin/ifup --force "$interface"
fi
```

Sleeper85, 2024-05-17:
The CAN bus status will be marked down after 20 loops (a different CAN ID is sent per loop) without response from the inverter.

[This number of IDs is different depending on the protocol chosen, with Deye I advise you to take "PYLON 1.2".](https://raw.githubusercontent.com/Sleeper85/esphome-jk-bms-can/main/images/CAN_Protocol_Table.png)

Deye only responds after receiving ID 0x356. **For me (at home) the links are turned OFF 1s every 2 hours** and I don't know why but that doesn't pose a problem. If the link was turned OFF due to non-response from the inverter it will be OFF for 120s so this problem does not come from my code.

After the link is marked down, **the code stops sending CAN IDs for 120s** before testing again for the presence of an inverter.

```ỲAML
interval:
  - interval: 120s
    then:
      - lambda: id(can_ack_counter) = 0; // Reset ACK counter for test inverter ACK
```

luckylinux, 2024-05-17:
In my case I am using PYLON+ Protocol (IIRC that was recommended a while ago by either you or MrPablo). Not sure if there are major Differences though ...

Sleeper85, 2024-05-17:
> In my case I am using PYLON+ Protocol (IIRC that was recommended a while ago by either you or MrPablo). Not sure if there are major Differences though ...

There is no point in using "PYLON +" with Deye because additional IDs are not supported by Deye. Just use the name "PYLON" with the protocol "PYLON 1.2".

luckylinux, 2024-05-18:
**IF** (when :smile:) I'll have another maintenance stop I'll switch the Protocol to Pylon 1.2.

While you are very likely right, I'm a bit scared of the system, given how much susceptible it was with this elusive bug I told you about (watchdog triggering if WiFi/MQTT/HA/... goes down) ...

Not necessarily an issue with the Code, there is probably some interaction going on between the different Components (and I probably have more Sensors enabled for tuning/troubleshooting than you do, so more RAM used, etc). 

The "fix" (workaround) as I said seems to increase the Watchdog Timeout to 30s (increasing to 10s improved the situation, but did NOT solve it), possibly combined with some of the other stuff I did (although since it was a Watchdog triggering a reboot, this is probably the solution: increasing the Watchdog Timeout). Why the ESP32 would "Freeze" / Hang and then trigger the Watchdog in the first Place, as I told you, I was not really able to diagnose.

Debug logs via USB showed everything normal then ... Watchdog Triggered ... Rebooting.

I'm more of the Attitude right now ... "If it works, don't touch it" :laughing: .

Sleeper85, 2024-07-24:
I also notice at home that the `CANBUS Status` goes from `ON` to `OFF` then `ON` in a few seconds every 2 hours, 4 hours, or more...

I have just reprogrammed the way to manage this status, the counter has been replaced by a `30s timer` (by default).

I'll see how it goes in the next few hours.

The two statuses to analyze are these:

[image]

cinusik, 2024-07-24:
> I also notice at home that the `CANBUS Status` goes from `ON` to `OFF` then `ON` in a few seconds every 2 hours, 4 hours, or more...
>
> I have just reprogrammed the way to manage this status, the [...]

@Sleeper85 I do not observe such behavior on 1.4.1 - it is running since 4 days without any problems. Do not see CANBUS going OFF and ON in HA history.

Sleeper85, 2024-07-24:
@cinusik 

> I do not observe such behavior on 1.4.1 - it is running since 4 days without any problems. Do not see CANBUS going OFF and ON in HA history.

So it may be linked to the brand of the inverter, we have a Deye @luckylinux  and I myself.

Sleeper85, 2024-07-24:
This may have something to do with the Deye `Time Of Use`. To investigate.

[image]
[image]

luckylinux, 2024-07-24:
@Sleeper85
Weird ... I probably have similar Settings but since I am not Grid Connected those are not enabled. Well neither are they enabled in your Case from what I see ...

But I guess you are right, there is definitively a correlation there, although I'd still say that only covers HALF the CANBUS Status Logs.

In my Case: 01h00, 05h00, 09h00, 13h00, 17h00, 21h00 ARE falling on the Time of Use Settings.

HOWEVER: 03h00, 07h00, 11h00, 23h00 are NOT falling on those times I would say ...

[image]
[image]
[image]

To me it looks more like a Periodic 2h (7200s) Thing. Maybe DHCP Related ? Inverter running some internal Calculations (e.g. calculating Energy Integrals, building Statistics, etc) ?

Sleeper85, 2024-07-24:
So several people with other inverters than Deye tell me they don't have this problem. This therefore seems linked to Deye and to an internal Deye timer which at a given moment does not send the canbus ACK within the allotted time of +/- 3s with the default parameters (30 loop of 100ms).

luckylinux, 2024-07-24:
Can you share the Version Details of your Deye ? I never upgraded my Firmware. I will probably upgrade soon, since there seems to be some benefits related to e.g. less Overcurrent Trips when doing a "blackstart" with Parallel Inverters, Fan ramping up/down, etc.

(That's what I was told at least, somewhere around this Thread: https://diysolarforum.com/threads/bonding-multiple-deye-12kw-inverters-in-off-grid-mode.70969/page-4#post-1071243)

Sleeper85, 2024-07-24:
Last upgrade a year ago, I contacted Deye support this morning to find out the current version and the changelog.

[image]

luckylinux, 2024-07-24:
Uhm alright. According to the User Group where I bought it (somebody put all Firmware files on Dropbox :smile:) these seems to be the latest:

```
MCU1-3PLV-Ver1001-C037-CK.bin
MCU2_SG12KLP3-V200x-1147-0614.bin
```

No Changelog info though.

Sleeper85, 2024-07-24:
> Uhm alright. According to the User Group where I bought it (somebody put all Firmware files on Dropbox 😄) these seems to be the latest for the 12kW Model:
> 
> ```
> MCU1-3PLV-Ver1001-C037-CK.bin
> MCU2_SG12KLP3-V200x-1147-0614.bin
> ```
> 
> No Changelog info though.

Do you also find the .bin for my model?

Deye never wanted to send them to me directly.

luckylinux, 2024-07-24:
It's only for the 12kW at the link I know ... Let me check if there is another one.

luckylinux, 2024-07-24:
Nope, even though the 6kW is currently being sold, the Group Guide is only (right now at least) pointing to a Firmware link for the 12kW. Or at least I think so, I don't know if the firmware is shared between Models (I do NOT think so).

I can try to ask though.

luckylinux, 2024-07-24:
Currently used Version, for Reference when I will upgrade later (maybe this Week):
[image]

luckylinux, 2024-07-24:
Got a Reply: "It is the same firmware file for all models under the SG04LP3 model name. I can just get it confirmed again by Deye."

I'll ask him to confirm with Deye ...

Sleeper85, 2024-07-25:
ESP32 flashing with new `canbus_link_timer` set to 3s (similar to the duration of the old 30 loop counter of 100ms).

[image]

[image]

Yesterday, from **12:49 to 00:00** no problems and then it starts again at **00:00**...
I'm going to reflash with a 5s timer to see how it goes.

[image]

Sleeper85, 2024-07-25:
With my PVbrain2 development board connected to a fake inverter sending a response exactly every 1s I do not notice this problem.

[image]

[image]

luckylinux, 2024-07-25:
@Sleeper85: I still got no confirmation from the Facebook Group.

I know Comparisons are difficult ... My data is still from the ~ 4 Month old - ish Branch.

Nevertheless I'd say your "Fake Inverter" just works better than the Deye :rofl:.

My Logs seems to show a pretty periodic Issue every 2h / almost 7200s. Although it's true that in Home Assistant I only have "Time of Use - Start" (anyway I do not use it since I am offgrid), so maybe the other "Events" (CANBus ON -> OFF -> ON) correspond to the "Time of Use - **End**".

It could be as simple as some Function/Routine/Interrupt in the Deye taking slighltly longer than it should (and a bad management of Priorities on their Behalf - BMS Disconnect should be pretty high on the list compared to an Energy Management Strategy, although of course lower than Protections such as Inverter Overcurrent/Short-Circuit/etc).

luckylinux, 2024-07-25:
@Sleeper85: I'm pretty sure some things happen at midnight. For once, all statistics about Solar Production (Daily PV Production, Daily Consumption, Daily Generator Production, Daily Export/Import, etc) are reset. So it could explain a "glitch" at midnight. Not the triggering of another Periodic (e.g. 2h-Period) Issue though. I'd expect a 24h Period for that Glitch (and again, bad management of sync/async Task and their Prioritities on the Deye Firmware Side).

**Some other Ideas ... Just brainstorming**

Just for my understanding (I'm not so much into it like you are): is it the Deye that sends you a Heartbeat and you "Reply" ("ACK") ? Or is it just you that sends a signal to the Deye in "Open Loop" ?

I'm wondering if there is also maybe some sync/async/interrupt "Issue" in your Code, but I believe that whenever there is some CANbus Data (or Heartbeat ecc), it's like a GPIO triggering an ISR (Interrupt Service Routine), so it's not like you are going to "miss" some of the Deye "sollicitations" because you are not having fully Synchronous Interrupts ?

What I mean is essentially: could it be that one Time your Function/Interrupt takes say 98 ms, sometimes 102 ms, sometimes 121 ms etc, and **sometimes** that "masks" some other incoming Data / Signal ? Basically kind of the equivalent in Power of a "Low Frequency Oscillation" introduced by a "High Frequency Signal with some small Time/Amplitude Variation or Jitter" ?

Did you try to measure how long your (high priority) ISR actually takes with a Scope (trigger GPIO pin HIGH-LOW-HIGH and measure Time Difference between the Rising and Falling Edges) and is it 100% Repeatable ? Or from time to Time (e.g. you calculate a Division, SQRT, whatever), it takes Longer ?

Sleeper85, 2024-07-25:
You are right, it is probably a problem on Deye's side and if increasing the timer to 5s hides this problem it will nevertheless still be there.

The PYLON canbus protocol stipulates that the inverter sends an ACK can_id `0x305` every `1s`.
My new `canbus_link_timer` waits `3s` before marking the canbus link `down`.

I can already tell you that Deye sends this ACK `0x305` only in response to the sending of can_id `0x356` and that it already does not respect the protocol.

[image]

luckylinux, 2024-07-25:
@Sleeper85: To be fair that Chinese-English Picture says "Inverter **reply** every second". Is that a Translation Error ?

Because "**Reply**" would mean that the BMS has to "Ping" the Inverter, NOT the other way around.

Sleeper85, 2024-07-25:
The inverter sends the can_id `0x305` every 1s as soon as it receives data from the BMS.
But in reality the Deye only sends it in response to the can_id `0x356` sent by the BMS.
The PYLON protocol does not specify any specific can_id to use for this heartbeat.

For my part, it's when this `0x305` frame is received that the 3s timer is reset.

I will try to analyze what I receive from Deye and check if it sends anything other than 0x305.

How do you explain that since yesterday 12:49 until today 00:00 the canbus has not been marked `down` once ?
Then since 00:00 it has been marked `down` every 2 hours.

So we can say that there are a few things that happen at 00:00, I know that the ESP also resets the BMS `Total Daily Energy` sensors at midnight.

luckylinux, 2024-07-25:
>> How do you explain that since yesterday 12:49 until today 00:00 the canbus has not been marked down once ? Then since 00:00 it has been marked down every 2 hours.

As I said I cannot :disappointed:. With your Modification, I can explain a ONE-OFF glitch at Midnight (00h00), but NOT what happened at 02h00, 04h00, 06h00, 08h00, ... It feels like whatever happens at Midnight (00h00) throws the entire System "Off-Balance".

One Thing that is surely weird with the Deye is that it has quite a high Time (Clock) Drift. IIRC I had like 20 Minutes Drift over 6 Months or so ... Not sure if this is related in any Ways though. The ESP32 is/might be NTP (or SNTP) Time Synchronized, but I do NOT think you give that information to the Deye, do you ? Then it's probably not related ...

Sleeper85, 2024-07-25:
I confirm that Deye only sends this `0x305` frame and nothing else.
The reception time varies between 1 and 3s.

Knowing that the ESP32 receives BMS data every `3s`, this adds a little time in the loop and probably delays the sending of can_id `0x356` and therefore the response `0x305` from the Deye ? I could work on this as well.

If the other inverters respond with can_id `0x305` to `each` can_id received then the heartbeat is surely shorter.

I have now increased the `canbus_link_timer` to `5s`.
Now we probably have to wait until midnight.
A little before midnight, I will check again the delay between two 0x305 ACKs.

I don't use the `BMS_Err_Stop` function currently, in the event of loss of links with the BMS the Deye will use the internal charging parameters that I have correctly defined.

[image]

[image]

luckylinux, 2024-07-25:
>> Knowing that the ESP32 receives BMS data every 3s, this adds a little time in the loop and probably delays the sending of can_id 0x356 and therefore the response 0x305 from the Deye ? I could work on this as well.

So you are currently only checking / pinging the CANbus on the Inverter side every 3s or so ? Like ... you only have 1 Loop in the Code ? Maybe Inverter CANbus communication should be done with another ISR/Loop triggered every 0.5s or 1.0s and configured with a higher priority.
Battery Communication might be another ISR (I guess you could do that ... we have 2 Cores on the ESP32-S3). Thoughts ?

>> I don't use the BMS_Err_Stop function currently, in the event of loss of links with the BMS the Deye will use the internal charging parameters that I have correctly defined.

So you are using the Battery in "Voltage Mode" (NOT Lithium) ? I know the Deye Inverter has this Good "Feature" that the BMS Communication works also when the Battery is in Voltage Mode.

Not sure if the Timing Changes (since mine is in Lithium and Warnings every 2h00 then I do NOT think it makes a difference).

Sleeper85, 2024-07-25:
I use Lithium 00 mode.

If the canbus link is up I send a different can_id every 100ms.

```YAML
  - interval: 100ms
    then:
      - if:
          condition:
            lambda: |-
              if (id(${canbus_id}_send_canbus_data) == true) {
                id(${canbus_id}_msg_counter)++;                            // CANBUS MSG counter ++
                return true;                                               // Condition OK
              }
              else return false;
```

I have to leave you, I have a team meeting now.

luckylinux, 2024-07-25:
No Worries :+1:. Just read my Posts when you have some Time. I'm just trying to give you some Ideas of what could be the Cause ...

Looking at the Code
[image]

You are doing:
- Reset `id(${canbus_id}_ack_counter)` every 120s (to prevent being "Stuck" in "OFF" I presume)
- Reset `id(${canbus_id}_ack_counter)` if you receive a Reply within 30 Cycles x 100ms = 3s (to prevent that each small Glitch accumulates over Time)

So those should be Fine.

Looking at your Picture, most "Steps" are 1s apart, some 3s, some 4s, some even 5s !

[image]

Those should actually trigger an Error. Why don't they ? You changed to 5s already ?
[image]
[image]

Question is, who is it at Fault ?
- Is it the Deye that is stuck in a low-priority ISR/Function and "Forgets" to reply ?
- Is it the ESP32 that is stuck in a low-priority ISR/Function and "Forgets" to send ?

It would be nice if you could print Timestamps with milliseconds, assuming it's possible to do so :+1:.

What I see in the Code (I could be mistaking) and what I see in the Timestamps from Home Assistant you provided seems to suggest the following:
a. Periodically, due to some Glight, you get `id(${canbus_id}_ack_counter) >= ${canbus_max_no_ack}` (well `=`, since it will NOT be increased after that anyways)
[image]

b. At some Point after that (impossible to know for sure but definitively < 120s), the reset happens since the Inverter sends an ACK finally:
[image]

It would be interesting if you could do the following for Debugging Purposes:
- Create a State/Entity "mapped" (assigning the value of) `id(${canbus_id}_ack_counter)`
- Keep increasing `id(${canbus_id}_ack_counter)` also in the other `else if` and `else` Blocks:
[image]

Is it possible that the CANBus Interrupts are Lower Priority than say UART/RS485/other stuff going on in your Code ? So basically that you are processing the CANbus signals only after everything else ?

On another Project with CANbus + Vertiv/Emerson R48 3000W Charger, the Issue I ran into was that the Code might fill up/flood the "Buffer" / "Queue" of the CAN Transceiver with Messages and nothing would go through anymore. Only solution would be to physically Reboot (and Power Cycle the USB Adapter). Not sure if something similar could be happening here with the ESP32. And maybe, once it goes to "OFF" State, then something forces a Reset of the CAN Adapter.

Possibly easier way to try "who is at fault": do a MINIMAL code (maybe even with just an USB Adapter if easier for you), just ACK of the Inverter and you sending some dummy Data, nothing else. Do the Timestamps exhibit the same "Pattern" (1s-2s-5s-1s-1s-1s, "all over the place"), or is it more Regular.

luckylinux, 2024-07-25:
Got a Reply (and Confirmation) from the Facebook Group:
```
**They confirm that it is the same file for all models in the SG04LP3 series**
```

Sleeper85, 2024-07-25:
@luckylinux 

The code you show now is completely different in YamBMS 1.4.2.

I just added a sensor with the `ACK interval` in ms.

[image]

Sleeper85, 2024-07-25:
Here is the new code since `YamBMS 1.4.2` and this did not prevent having the same problem every 2 hours since midnight. I just added the new `ACK interval` sensor.

```CPP
# +--------------------------------------+
# | Scripts                              |
# +--------------------------------------+
script:
  - id: ${canbus_id}_script_canbus_link_timer
    mode: restart
    then:
      - delay: ${canbus_link_timer}s
      - lambda: |-
          id(${canbus_id}_send_canbus_data) = false;                      // Stop sending CANBUS datas
          id(${canbus_id}_status).publish_state(false);                   // Set CANBUS Status to OFF

# +--------------------------------------+
# | Start CAN Handling                   |
# +--------------------------------------+

canbus:
  # 0x305 - Inverter ACK - SMA/LG/Pylon/Goodwe reply
  - id: !extend ${canbus_node_id}
    on_frame:
    - can_id: 0x305
      then:
        - light.toggle:
            id: ${canbus_light_id}
        - lambda: |-
            id(${canbus_id}_send_canbus_data) = true;                      // Continue sending CANBUS datas
            id(${canbus_id}_status).publish_state(true);                   // Set CANBUS Status to ON
            id(${canbus_id}_script_canbus_link_timer).execute();           // Restart CANBUS link timer

            // ACK interval
            int current_ack_ms = millis();
            int ack_interval_ms = (current_ack_ms - id(${canbus_id}_last_ack_ms));
            id(${canbus_id}_last_ack_ms) = current_ack_ms;
            id(${canbus_id}_ack_interval_ms).publish_state(ack_interval_ms);
            ESP_LOGI("canbus", "can_id 0x305 received - ack_interval %d ms", ack_interval_ms);

interval:
  # Test inverter 0x305 ACK
  - interval: 120s
    then:
      - lambda: |-
          if (id(${canbus_id}_send_canbus_data) == false) {
            id(${canbus_id}_send_canbus_data) = true;                      // Start sending CANBUS datas
            id(${canbus_id}_script_canbus_link_timer).execute();           // Restart CANBUS link timer
          }

  - interval: 100ms
    then:
      - if:
          condition:
            lambda: |-
              if (id(${canbus_id}_send_canbus_data) == true) {
                id(${canbus_id}_msg_counter)++;                            // CANBUS MSG counter ++
                return true;                                               // Condition OK
              }
              else return false;

          then:
```

Sleeper85, 2024-07-25:
Wait and see what happens at midnight.

[image]

luckylinux, 2024-07-25:
So you just flashed with 5s "timeout" now ? It doesn't look so bad ... Compare that with your previous Picture where it could be up to 4s-5s (between 2 consecutive ACK) ...

Is this with the "Full Code" or just the "Minimal Example" to try to isolate whether the Issue is on Deye or ESP32 Side ?

Sleeper85, 2024-07-25:
Yes I flashed the complete code with a `canbus_link_timer` of `5s`.

The interval has just increased to `2.91s`, therefore close to the `3s` of the first test.

[image]

Sleeper85, 2024-07-25:
> Donc tu viens de flasher avec un "timeout" de 5 secondes maintenant ? Cela n'a pas l'air si mal... Comparez cela avec votre image précédente où cela pourrait aller jusqu'à 4s-5s (entre 2 ACK consécutifs)...

This shows that we cannot trust the timestamp of the logs.

luckylinux, 2024-07-25:
> > Donc tu viens de flasher avec un "timeout" de 5 secondes maintenant ? Cela n'a pas l'air si mal... Comparez cela avec votre image précédente où cela pourrait aller jusqu'à 4s-5s (entre 2 ACK consécutifs)...
> 
> This shows that we cannot trust the timestamp of the logs.
Is GitHub autotranslating all messages :open_mouth: ?

Si tu veux on peut aussi bien parler Français :laughing:

luckylinux, 2024-07-25:
> The interval has just increased to `2.91s`, therefore close to the `3s` of the first test.

And it's close to the 2h - "integer" Period (12h00, 14h00, 16h00, ...).

I would have thought the Deye would compute hour-by-hour Statistics, NOT bi-hourly (every 2 Hours).

Are you (or the ESP32  by itself :laughing:) doing something on the ESP32 side every 2 Hours ? Maybe something to do with DHCP / NTP / SNTP Sync ?

luckylinux, 2024-07-25:
@Sleeper85: can you provide a Screenshot of your "Time of Use Display" ?

Maybe a "Workaround" (if you do NOT use Time of Use) could be to set every Time to "00:00". And see if the Issue (or "Feature") disappears ...

Sleeper85, 2024-07-25:
> Are you (or the ESP32 by itself 😆) doing something on the ESP32 side every 2 Hours ? Maybe something to do with DHCP / NTP / SNTP Sync ?

Good question, I don't know.

I'm testing it with the PVbrain2 board and a fake inverter which sends an ACK `0x305` only in response to the BMS can_id `0x356` as is the case with Deye.

`canbus_link_timer` is set to `3s` in order to see if disconnections happen or not?

Wait and see

luckylinux, 2024-07-25:
In my case changing the Time of Use settings doesn't seem to affect it (done Remotely, didn't Test it, but setting seem to be kept at the set value of 00:00). I got another error just now at 15h03.

[image]

Strange though that I you get "even" Numbers while I get "odd" Numbers, eh ?

Sleeper85, 2024-07-25:
It's 16:00, time to take a little `3.10s` break :smile: 

This confirms that this behavior comes from the Deye inverter and not from the ESP32.

[image]

luckylinux, 2024-07-25:
I think you might need to report a BUG to Deye to be honest. Unless I need to flash my Firmware and update my Inverters first to test :roll_eyes:  ...

Sleeper85, 2024-07-25:
> Je pense que vous devrez peut-être signaler un BUG à Deye pour être honnête. À moins que je doive d'abord flasher mon Firmware et mettre à jour mes onduleurs pour tester 🙄...

This will be much too complicated to explain to them and as you said, from 00:00 and every 2 hours Deye carries out a task which slightly delays the sending of ACK 0x305. So in theory with the `canbus_link_timer` parameter set to `5s` the links should be stable.

luckylinux, 2024-07-25:
Well you can argue it's just a false alarm and could be filtered.

From a "tripline" point of view, it's anyway the Inverter that will trigger and Error and shut-down (BMS_Err_Stop) in case there is a real Communication Issue. 

And that will NOT be affected by 1s, 3s or 5s "Internal" ESP32 (to your Code) CANbus Supervision.

I would leave the "ACK" Entity (~ 1s) in case there is a real issue where the Deye Trips for whatever reason and we need to investigate. Of course if it doesn't use too much Memory or CPU.

But yeah, you can "filter" the CANbus status for these false alarms that did NOT trip the Inverter.

Sleeper85, 2024-07-25:
This is already the case and I will keep this new sensor in YamBMS 1.4.2. This is interesting for debugging.

```ỲAML
sensor:
  # 0x305 ACK interval
  - platform: template
    id: ${canbus_id}_ack_interval_ms
    name: "${name} ${canbus_name} Inverter ACK interval"
    update_interval: 1s
    accuracy_decimals: 0
    unit_of_measurement: ms
    icon: mdi:update
    entity_category: diagnostic
```

luckylinux, 2024-07-25:
>> This is already the case
Missed that Part, Sorry :disappointed:. You were already thinking further ahead :+1:.

luckylinux, 2024-07-25:
@Sleeper85: given what happened at midnight last Time (~ 18 hours ago), I'd be curious to see what the "Inverter ACK Interval" will show at Midnight (if it "jumps" higher to say 5 Seconds). Or if these ~ 3 Seconds are "constant" (every "integer" even/odd Hour).

Sleeper85, 2024-07-25:
> @Sleeper85: given what happened at midnight last Time (~ 18 hours ago), I'd be curious to see what the "Inverter ACK Interval" will show at Midnight (if it "jumps" higher to say 5 Seconds). Or if these ~ 3 Seconds are "constant" (every "integer" even/odd Hour).

14:00 = 2.91s
16:00 = 3.10s
18:00 = 3.24s

luckylinux, 2024-07-25:
Yeah it's approx. 3s now. I mean, what it will be at 00:00 :rofl:.

Sleeper85, 2024-07-25:
> Yeah it's approx. 3s now. I mean, what it will be at 00:00 🤣.

14:00 = 2.91s
16:00 = 3.10s
18:00 = 3.24s
**20:00 = 2.26s**

luckylinux, 2024-07-25:
That drop is quite weird ...

Are you sending the "Ping" `0x356` every 1s or more often like I suggested ?

Assuming the Deye always takes the same amount of Time (**BIG** Assumption) you should observe a distribution with a width of 1s (or whatever often you are sending it).

So far it's around that 1s Width (min = 2.26 s , max = 3.24s).

Sleeper85, 2024-07-25:
> Are you sending the "Ping" `0x356` every 1s or more often like I suggested ?

I could easily send it twice as much as the others.

Sleeper85, 2024-07-25:
I just understood why the time of a canbus loop is `1.20s`.

I send the IDs in a precise order, my counter `${canbus_id}_msg_counter` goes up to `12` before being reset when sending the last can_id `0x35E` which is the name of the BMS and sent regardless the protocol.

12 x 100ms = 1.2s

Regardless of the protocol chosen, you must do 12 loops of 100ms to complete a cycle.

luckylinux, 2024-07-25:
> > Are you sending the "Ping" `0x356` every 1s or more often like I suggested ?
> 
> I could easily send it twice as much as the others.

Well ... if it's NOT you that sends the "Ping" every 1s (or 1.2s), as you said you send it much more often than that (every 100ms), maybe it's something internal to the Deye that is responsible for the majority of the Delay. So even at twice the frequency / half the "ping" Period, if the Deye already was responsible for most of the Delay, we might only gain a few ms, not much more.

Basically if the Deye internal control loop is "tuned" to (approx.) 1s based on PYLON Protocol and the 0.2s is the "overhead" that goes on top of that.

So ... If you try to send twice as often but it only drops to 1.1s (during normal use) or 2.9s (during "integer" hours: 02h00, 04h00, ...), then it's just something internal to the Deye I'd say.

@cinusik: what is you Brand of Inverter and Protocol Used ?

luckylinux, 2024-07-25:
> I just understood why the time of a canbus loop is `1.20s`.
>
> I send the IDs in a precise order, my counter `${canbus_id}_msg_counter` goes up to `12` before being reset when sending the last can_id [...]

Is that an implementation "Feature" (easier from a Coding perspective) or is it "Needed" to be like that actually ?

When you say the `can_id` (I'm not a CANbus Expert), do you basically mean the CANbus "Message Content" (in Layman's Terms) ? Beacuse when I hear ID I mostly think about Modbus/RS485 "Device ID" or "Device Address".

So basically you do 12 "send" of 12 different Messages (each of them being one of: Battery Voltage, Battery Temperature, Allowed Current, SOC, Name of BMS, etc).

Sleeper85, 2024-07-25:
When I talk about can_id, these are the messages sent to the inverter see [the table in the README](
https://github.com/Sleeper85/esphome-jk-bms-can/tree/development?tab=readme-ov-file#can-protocol-table).

I think I will be able to reduce this canbus loop time a little.

[94 more replies]

closed 2024-08-02 by Sleeper85