Sleeper85/esphome-jk-bms-can #49: Balance is reported as ON while actually beeng OFF

virus100b, 2024-05-15:
Hello. The problem i have is that sometimes is reporting balance ON when actually balance is OFF, checked by connecting to the BMS over bluetooth. If i restart the ESP the Balance reporting is showing correct state. I don't know why is doing that.

ESP Home - Balancing reported as ON.
[image]

JK BT app. Balancing is OFF.
[image]

virus100b, 2024-05-15:
Same behavior related to Charging Switch.

ESP Home Charging switch reported as Off.
[image]

BMS App - Charging is ON
[image]

Sleeper85, 2024-05-18:
Several people including @MrPabloUK have reported a problem refreshing data received when logger is on INFO. This problem does not occur with logger on DEBUG. We don't know if this is related to a certain version of HA, esphome or syssi components, this still needs to be verified.
Please change your config and stay with DEBUG in production.

```YAML
logger:
  level: DEBUG
```

virus100b, 2024-05-18:
I will change that and monitor. Thank you !

virus100b, 2024-05-20:
I have modified logger to DEBUG and now is working fine. 
Thank you for help!

closed 2024-07-24 by Sleeper85