Sleeper85/esphome-jk-bms-can #53: DEV: some "by default" values wrong for LiFePo4

txubelaxu, 2024-06-13:
Some "by default" values are wrong for LiFePo4 batteries:
[image]

Sleeper85, 2024-06-13:
If the conditions are not met for the combination of BMS, the default value of (MIN) value is 5V and (MAX) value 0V. The BMS compares its own value against the default value. In your case it seems that the BMS information is not combined?

txubelaxu, 2024-06-14:
The problem is that the info from BMSs goes from valid values to NA, then to valid values, then to NA...
[image]
[image]

I have to check this issue in the sniffer/bms code. But, anyway... some ideas:
* The BMS info can be of three types: status, settings and "other info". I do not know how frequently the sniffer must ask for each type of info. The one that changes almost always is the first one (status). The other two info normally does not change, but sniffer must ask for it because there are another ways to change the values (bluetooth...). I think is not correct to pass this settings info to NA when missing answer for a short time.
* 5,00V is a very wrong value for LiFePo4 cells. 3,65V is the physical limit. 3,61V<overvoltage protection<3,65V
* 0,00V is a very wrong value for LiFePo4 cells. 2,6V is a good limit for undervoltage protection.

As said, I have to check if this issue is because a bug in my code.

Sleeper85, 2024-06-14:
0V and 5V are the default values ​​of the sensors when no BMS is combined. I know this is very far from the real values ​​for LFP but we also support Li-ion with a bulk charging voltage of 4.9V. In production you should never see these values, if you see them it is because none of your BMS are combined. In addition, this has no impact on the BMS or the inverter nor sent on the CAN bus.

virus100b, 2024-06-15:
I just made a test. I have set the same address  (01) for BMS1 in BMS2 in Yaml code trying to trick the ESP to see 2 BMS-s as i just have just one for now. I can see the entities values in esphome BMS1 and 2 identical of course but still Smartbms is reporting UVP, OVP and UTP alarms.

Smart BMS 1 Alarm OVP
Smart BMS 1 Warning UTP

What are the condition to met in order to combine? 
Looks like i cannot run just one BMS with this code. What i can do to run just one bms?

Sleeper85, 2024-06-16:
I just found a block of code that I forgot to remove and which may have created disturbances and I improved the way of publishing the MIN/MAX values ​​of the cells. I'll test it then publish the new code.

@virus100b It is entirely possible to use my code with a single BMS, however I don't know if this is a problem for @txubelaxu  code.

virus100b, 2024-06-16:
> I just found a block of code that I forgot to remove and which may have created disturbances and I improved the way of publishing the MIN/MAX values ​​of the cells. I'll test it then publish the new code.

Great. We are waiting.

txubelaxu, 2024-06-16:
It is not a problem having a only one BMS when using the esp32 as master
(i.e. mode2)

Sleeper85, 2024-06-16:
The new code is published.

[Regarding refreshing the sensors, please keep in mind that some people have encountered problems with logger level INFO.](https://github.com/Sleeper85/esphome-jk-bms-can/issues/49)
I don't know if this problem comes from the syssi code, the esphome version or HA.

So I advise you to keep logger level DEBUG (default value) as below.

```YAML
logger:
```

Sleeper85, 2024-06-16:
> What are the condition to met in order to combine?

All these conditions must be met:
- BMS switch charging ON
- BMS switch discharging ON
- BMS no errors
- Combined switch ON

```YAML
switch:
# Combine Enable-Switch (user controlled switch to en/dis-able combine)
  - platform: template
    name: "${name} ${bms_name} Combine enabled"
    id: bms${bms_id}_switch_combine
    restore_mode: RESTORE_DEFAULT_ON
    optimistic: true

binary_sensor:
# Combine Availability (indicated, if the device is able to be included in combine)
  - platform: template
    name: "${name} ${bms_name} Combine Availability"
    id: bms${bms_id}_availability
    lambda: return ((id(bms${bms_id}_switch_charging).state) && (id(bms${bms_id}_switch_discharging).state) && (id(bms${bms_id}_errors_bitmask).state == 0));

# Combine Status (indicated, if the device is included in combine)
  - platform: template
    name: "${name} ${bms_name} Combine"
    id: bms${bms_id}_combine
    lambda: return ((id(bms${bms_id}_availability).state) && (id(bms${bms_id}_switch_combine).state));
```

Sleeper85, 2024-06-16:
The `Smart BMS 1 Battery module` sensor returns the number of BMS currently combined.
The only way to see min/max values ​​like 0V or 5V is if the combined BMS count is `0`.

So during your tests `Smart BMS 1 Battery module` must be at least equal to `1`.

txubelaxu, 2024-06-16:
Well, I think I have working.
I have not activated web and I have set this way:

      bms_update_interval: '30s' # going below '3s' can cause problems
      bms_combine_interval: '10s'

virus100b, 2024-06-16:
@Sleeper85  Good news! 
I have activated charge and discharge on my BMS (Remotely, using my whife :)), as the ESPHome switches ar still not changing the state and i am not home)
I have update the new files and flashed.
Now i can see Smart BMS 1 Battery module = 2 (i still have cloned  BMS 2 reading the adrress 01  to simulate 2 BMS)
All is looks to work fine 
Smart BMS 1 Battery Capacity Total (Σ) = 560 Ah (2 x 280Ah)
Smart BMS 1 Max Charge current (Σ) = 300 A (Each BMS has 150A set value)

The only thing that is  not displayed correctly now is power tube - mos temperature protection and recovery sensors, and  operation status.

[image]
As you can see the esp is reading MOS temperature protection and recovery correctly but  the code is looking for power tube protection and recovery names.
For the sensor named JK-BMS 1 operation status i don't know from  where is reading the state.

Another thing..... I don't know if anybody notice is:
JK-BMS 1 temperature powertube temperature is the integer value of  JK-BMS 1 temperature sensor 3 which is the MOS temperature.
[image]

Actually the temperatures displayed in ESPHome are:
smartbms JK-BMS 1 temperature sensor 1 =  temperature sensor probe 1 (Battery T1 in BMS)
smartbms JK-BMS 1 temperature sensor 2 =  temperature sensor probe 2 (Battery T2 in BMS)
smartbms JK-BMS 1 temperature sensor 3 =  MOS temp in BMS
smartbms JK-BMS 1 temperature sensor 4 =  temperature sensor probe 3 (Battery T4 in BMS)
smartbms JK-BMS 1 temperature sensor 5 =  temperature sensor probe 4  (Battery T5 in BMS)

I guess  JK-BMS 1 temperature sensor 3 need be renamed to MOS temp and power tube sensor can be removed.

virus100b, 2024-06-16:
> The new code is published.
>
> [Regarding refreshing the sensors, please keep in mind that some people have encountered problems with logger level [...]

Yes. I had that problem with the code for JK B series BMS but was fixed after i have changed the logger to debug as you advise. 
logger:
  level: DEBUG
No issues since then.

txubelaxu, 2024-06-16:
The PDF shows 5 temperatures:

[image]

[image]

I do not know if one of them is repeated

virus100b, 2024-06-16:
I have a new BMS (V15.17) and mine have just 4 probes but the app is showing the names for 5 temp. 
You can check in your app and see if is the same like mine. In my case T3 from ESPHome is MOS temp with 0.1 decimal as you can see in the picture that i have attached.  ESP MOS temp is the integer value of T3.

txubelaxu, 2024-06-16:
Yes, @virus100b. Here it is the same like yours. And ESP MOS temp is the integer value of T3 here as well.

txubelaxu, 2024-06-16:
Now it seems stable here:

[image]

virus100b, 2024-06-16:
> integer value of T3 here as well.

Then probably BMS is outputting the MOS temp twice: First on address  0x128A (integer value) and on 0x12F8 (the value with one decimal).  This is not an error is just a fact that i have discovered.

txubelaxu, 2024-06-16:
> > integer value of T3 here as well.
> 
> Then probably BMS is outputting the MOS temp twice: First on address 0x128A (integer value) and on 0x12F8 (the value with one decimal). This is not an error is just a fact that i have discovered.

That is why is not showed T3 in the app.

virus100b, 2024-06-16:
> That is why is not showed T3 in the app.
Yes.

Is strange that they have chose T3 for MOS. It was more logical to chose T5 and let T1-T4 for probes sensors.

txubelaxu, 2024-06-16:
Yes, and I think there is something strange in the wiring as well.

txubelaxu, 2024-06-16:
Bad news. My esp32 continues restarting. :-( 
I think it is because some overflow or bug feom sniffer or bms.

cinusik, 2024-06-16:
@txubelaxu I experienced the same during testing. RS485 connection was unstable than restarts every few minutes.

txubelaxu, 2024-06-16:
Thanks for the confirmation, @cinusik. There is a bug that cannot find. I think is an overflow somewhere.

virus100b, 2024-06-17:
I have no trouble with restart. I am using a ESP32 C3 super mini board.
[image]

 I know that you may have  restart issues if the  is not connecting to HA.

# Please use the native `api` component instead of the `mqtt` section.
# If you use Home Assistant, the native API is more lightweight.
# If there is no HA server connected to this API, the ESP32 reboots every 15 minutes to try to resolve the problem.
# If you don't use Home Assistant please uncomment the "reboot_timeout: 0s" option.
api:
  reboot_timeout: 0s

Sleeper85, 2024-06-17:
@txubelaxu @cinusik

What board do you use?

I think @widget4145 has no reboot issues with Atom S3 Lite (ESP32-S3) and 7x JK-BMS PB series. 

Classic ESP32 is less powerful than ESP32-S3 or ESP32-C3.

Sleeper85, 2024-06-17:
In addition, the Web server uses a lot of additional resources and may be the cause of the reboot problem. 

`Please note that enabling this component will take up a lot of memory and may decrease stability, especially on ESP8266.`

https://esphome.io/components/web_server.html

txubelaxu, 2024-06-17:
I am using esp32doit-devkit-v1 config. Hardware is: NodeMCU-ESP32 ESP32 DEVKITV1
I have disabled web server and disabled the logging as well.

[image]

It restarts continuously. :-(

cinusik, 2024-06-17:
@virus100b @Sleeper85 As stated on the DIY forum, I'm using AtomS3, web disabled, logger is DEBUG. There is a connection to HA. BMSes randomly show up and disappear and it restarts from time to time irregularly. Ordered shielded communication cables for RS485 and CAN and will do more testing then.

txubelaxu, 2024-06-17:
Hi @cinusik. I see that my jk_rs485_bms.cpp is causing those restarts. I am trying to debug but it is something random, so I am trying changing the code.

cinusik, 2024-06-17:
@txubelaxu OK, waiting to test it once done.

txubelaxu, 2024-06-17:
I am testing esphome-jk-bms@txubelaxu

The problem is localized at the jk_rs485_bms.cpp and forward. I have supposed that was a limit of the size of json (512). I have imported the json component into my project and changed from 512 to 1024. But the problem persists in the same point. When arrives that point and most of the times nothing happens, but randomly it hangs and restarts. Always at the same point (randomly):

[10:19:36][V][sensor:043]: 'rs485-bms1 battery total runtime': Received new state 12602836.000000
[10:19:36][D][sensor:094]: 'rs485-bms1 battery total runtime': Sending state 12602836.00000 s with 0 decimals of accuracy
[10:19:36][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[10:19:36][V][json:058]: Size after shrink 124 bytes

[10:22:57][V][sensor:043]: 'rs485-bms1 battery total runtime': Received new state 12603038.000000
[10:22:57][D][sensor:094]: 'rs485-bms1 battery total runtime': Sending state 12603038.00000 s with 0 decimals of accuracy
[10:22:57][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[10:22:57][V][json:058]: Size after shrink 124 bytes

[10:24:17][V][sensor:043]: 'rs485-bms1 battery total runtime': Received new state 12603117.000000
[10:24:17][D][sensor:094]: 'rs485-bms1 battery total runtime': Sending state 12603117.00000 s with 0 decimals of accuracy
[10:24:17][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[10:24:17][V][json:058]: Size after shrink 124 bytes

[10:26:37][V][sensor:043]: 'rs485-bms1 battery total runtime': Received new state 12603258.000000
[10:26:37][D][sensor:094]: 'rs485-bms1 battery total runtime': Sending state 12603258.00000 s with 0 decimals of accuracy
[10:26:37][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[10:26:37][V][json:058]: Size after shrink 124 bytes

[10:32:04][V][sensor:043]: 'rs485-bms1 battery total runtime': Received new state 12603585.000000
[10:32:04][D][sensor:094]: 'rs485-bms1 battery total runtime': Sending state 12603585.00000 s with 0 decimals of accuracy
[10:32:04][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[10:32:04][V][json:058]: Size after shrink 124 bytes

[10:48:08][V][jk_rs485_bms:448]: PRECHARGE WORKING STATUS: 0x00
[10:48:08][D][jk_rs485_bms:451]: UserAlarm: 0x00 0x00 (always 0xC5 0x09?)
[10:48:08][V][sensor:043]: 'rs485-bms1 battery total runtime': Received new state 12604549.000000
[10:48:08][D][sensor:094]: 'rs485-bms1 battery total runtime': Sending state 12604549.00000 s with 0 decimals of accuracy

[11:08:05][V][sensor:043]: 'rs485-bms1 battery total runtime': Received new state 12605746.000000
[11:08:05][D][sensor:094]: 'rs485-bms1 battery total runtime': Sending state 12605746.00000 s with 0 decimals of accuracy

[11:16:52][V][jk_rs485_bms:452]:        1] temp_param_value: 12606272.000000
[11:16:52][V][sensor:043]: 'rs485-bms1 battery total runtime': Received new state 12606272.000000
[11:16:52][D][sensor:094]: 'rs485-bms1 battery total runtime': Sending state 12606272.00000 s with 0 decimals of accuracy
[11:16:52][V][json:038]: Attempting to allocate **512** bytes for JSON serialization
[11:16:52][V][json:058]: Size after shrink 124 bytes

[11:19:11][V][jk_rs485_bms:452]:        1] temp_param_value: 12606413.000000
[11:19:11][V][sensor:043]: 'rs485-bms1 battery total runtime': Received new state 12606413.000000
[11:19:11][D][sensor:094]: 'rs485-bms1 battery total runtime': Sending state 12606413.00000 s with 0 decimals of accuracy
[11:19:11][V][jk_rs485_bms:454]:        2] temp_param_value: 12606413.000000

[12:11:05][V][jk_rs485_bms:452]:        1] temp_param_value: 12609525.000000
[12:11:05][V][sensor:043]: 'rs485-bms1 battery total runtime': Received new state 12609525.000000
[12:11:05][D][sensor:094]: 'rs485-bms1 battery total runtime': Sending state 12609525.00000 s with 0 decimals of accuracy
[12:11:05][V][json:038]: Attempting to allocate **1024** bytes for JSON serialization
[12:11:05][V][json:058]: Size after shrink 124 bytes
[12:11:05][V][jk_rs485_bms:454]:        2] temp_param_value: 12609525.000000

virus100b, 2024-06-17:
> @virus100b @Sleeper85 As stated on the DIY forum, I'm using AtomS3, web disabled, logger is DEBUG. There is a connection to HA. BMSes randomly show up and disappear and it restarts from time to time irregularly. Ordered shielded communication cables for RS485 and CAN and will do more testing then.

That is strange. Should not have any issues with that hardware. 

Did you measure if you have 60Ohm between your RS485 A+ and B+? (ESP and BMS must be OFF to be able to measure)
I had a issue with a ttl to rs 485 converter board where the termination resistor from the board was out of specs and the communication was  throwing eronate values. I have replaced the resistor with a 120Ohm and never had issues.
Check that an see if you have 60 Ohm with TTL- RS485 converter connected to the BMS or you can test independently but in this case you need to read 120Ohm on the  BMS and 120 On the converter

txubelaxu, 2024-06-17:
It is very strange. After last changes, In VERY_VERBOSE I have not seen failing. But in VERBOSE mode yes.

Maybe it will fail soon. :sob:

txubelaxu, 2024-06-17:
Almost 3 hours without restarting from here.

txubelaxu, 2024-06-19:
Debugging... I have found where is the problem. But I do not know how to solve it.
in jk_rs485_bms are getting the values and publishing them

```
  bool value_tmp;
  value_tmp=this->check_bit_of_byte_(data[282], 0); 
  ESP_LOGI(TAG, "  Getting bit 0 from value: %d is %02X of object with address %p",data[282],value_tmp,(void *) this->heating_switch_);   
  this->publish_state_(this->heating_switch_, value_tmp);
  // ESP_LOGI(TAG, "  heating switch: %s", ( this->check_bit_(data[282], 1)) ? "on" : "off");
  value_tmp=this->check_bit_of_byte_(data[282], 1); 
  ESP_LOGI(TAG, "  Getting bit 1 from value: %d is %02X of object with address %p",data[282],value_tmp,(void *) this->disable_temperature_sensors_switch_);   
  this->publish_state_(this->disable_temperature_sensors_switch_, value_tmp);
  value_tmp=this->check_bit_of_byte_(data[282], 2); 
  ESP_LOGI(TAG, "  Getting bit 2 from value: %d is %02X of object with address %p",data[282],value_tmp,(void *) this->gps_heartbeat_switch_);   
  this->publish_state_(this->gps_heartbeat_switch_, value_tmp); 
  value_tmp=this->check_bit_of_byte_(data[282], 3); 
  ESP_LOGI(TAG, "  Getting bit 3 from value: %d is %02X of object with address %p",data[282],value_tmp,(void *) this->port_selection_switch_);   
  this->publish_state_(this->port_selection_switch_, value_tmp);
  // ESP_LOGI(TAG, "  Port switch: %s", this->check_bit_(data[282], 8) ? "RS485" : "CAN");
  value_tmp=this->check_bit_of_byte_(data[282], 4); 
  ESP_LOGI(TAG, "  Getting bit 4 from value: %d is %02X of object with address %p",data[282],value_tmp,(void *) this->display_always_on_switch_);  
  this->publish_state_(this->display_always_on_switch_, value_tmp);
  value_tmp=this->check_bit_of_byte_(data[282], 5); 
  ESP_LOGI(TAG, "  Getting bit 5 from value: %d is %02X of object with address %p",data[282],value_tmp,(void *) this->special_charger_switch_);   
  this->publish_state_(this->special_charger_switch_, value_tmp);
  value_tmp=this->check_bit_of_byte_(data[282], 6); 
  ESP_LOGI(TAG, "  Getting bit 6 from value: %d is %02X of object with address %p",data[282],value_tmp,(void *) this->smart_sleep_on_switch_);  
  this->publish_state_(this->smart_sleep_on_switch_, value_tmp);
  value_tmp=this->check_bit_of_byte_(data[282], 7); 
  ESP_LOGI(TAG, "  Getting bit 7 from value: %d is %02X of object with address %p",data[282],value_tmp,(void *) this->disable_pcl_module_switch_);  
  this->publish_state_(this->disable_pcl_module_switch_, value_tmp);
```
The problem is that the address of the pointer this->disable_pcl_module_switch_ sometimes changes:
[12:16:12][V][jk_rs485_bms:1138]: Publishing state 0 for object with address 0x3ffc958c
[12:16:12][V][jk_rs485_bms:1138]: Publishing state 0 for object with address 0x3ffc8fb4
[12:16:12][V][jk_rs485_bms:1138]: Publishing state 0 for object with address 0x3ffc9848
[12:16:12][V][jk_rs485_bms:1138]: Publishing state 0 for object with address 0x3ffc98a0
[12:16:12][V][jk_rs485_bms:1138]: Publishing state 1 for object with address 0x3ffc94dc
[12:16:12][V][jk_rs485_bms:1138]: Publishing state 0 for object with address 0x3ffc98f8
[12:16:12][V][jk_rs485_bms:1138]: Publishing state 0 for object with address 0x3ffc9740
[12:16:12][V][jk_rs485_bms:1138]: Publishing state 0 for object with address **0x3ffc9798**
[12:16:13][I][jk_rs485_bms:886]:   After binary 7 -------------------------------------------------------------------------------------

[12:16:22][I][jk_rs485_bms:853]:   Before binary 0 -------------------------------------------------------------------------------------------
[12:16:22][V][jk_rs485_bms:1138]: Publishing state 0 for object with address 0x3ffc958c
[12:16:22][V][jk_rs485_bms:1138]: Publishing state 0 for object with address 0x3ffc8fb4
[12:16:22][V][jk_rs485_bms:1138]: Publishing state 0 for object with address 0x3ffc9848
[12:16:22][V][jk_rs485_bms:1138]: Publishing state 0 for object with address 0x3ffc98a0
[12:16:22][V][jk_rs485_bms:1138]: Publishing state 1 for object with address 0x3ffc94dc
[12:16:22][V][jk_rs485_bms:1138]: Publishing state 0 for object with address 0x3ffc98f8
[12:16:23][V][jk_rs485_bms:1138]: Publishing state 0 for object with address 0x3ffc9740
[12:16:23][V][jk_rs485_bms:1138]: Publishing state 0 for object with address **0x8faa**

When this happens, the esp32 hangs and restarts.

Somebody knows what could be the reason and how to solve?

cinusik, 2024-06-19:
@txubelaxu That explain my random restarts like after 16 minuter, 28 minutes, 2 hours etc. Great you found this! But not so great without knowing the reason :( Maybe some bug on JK firmware side? Asking them will take ages waiting for answer, if any.

txubelaxu, 2024-06-19:
No jk company bug.
It is a code bug sure. My code or esphome bug.

cinusik, 2024-06-19:
@txubelaxu OK, understand :) Just for testing purposes can you ignore this PCL module state in your code so it will be not published and not causing ESP to hang?

txubelaxu, 2024-06-19:
Hi @cinusik. I have uploaded now with that line commented.
```  value_tmp=this->check_bit_of_byte_(data[282], 7); 
  ESP_LOGI(TAG, "  Getting bit 7 from value: %d is %02X of object with address %p",data[282],value_tmp,(void *) this->disable_pcl_module_switch_);  
  ////this->publish_state_(this->disable_pcl_module_switch_, value_tmp);```

But I am printing the value of the address. I cannot understand WHY
1. only happens with disable_pcl_module_switch_ and not to previous all switch_ lines?
2. if I comment that line the bug dissapear?... so it is not printed the uncorrect address in the previous line!!!

I cannot believe it.

Sleeper85, 2024-06-19:
I'm far from being an expert in creating ESPhome components but maybe @syssi could help?

txubelaxu, 2024-06-19:
Corrected: is happening as well. The address of this->disable_pcl_module_switch_ changes
```
  value_tmp=this->check_bit_of_byte_(data[282], 7); 
  ESP_LOGI(TAG, "  Getting bit 7 from value: %d is %02X of object with address %p",data[282],value_tmp,(void *) this->disable_pcl_module_switch_);  
  ////this->publish_state_(this->disable_pcl_module_switch_, value_tmp);
```

txubelaxu, 2024-06-20:
I'm going crazy. It only happens with the parameter called "disable_pcl_module_switch_". I've tried to fix the issue by reordering the code to place it higher up. I've also tried to fix the problem by commenting out the rest of the switch parameters that are obtained from data[282]. It only happens with the address of disable_pcl_module_switch of the second BMS. Its address changes randomly.

```
[12:08:26][I][jk_rs485_bms:859]: [01] heating_switch_                      is bit 0 of 16 is 00 address 0x3ffc8f58
[12:08:26][I][jk_rs485_bms:863]: [01] disable_temperature_sensors_switch_  is bit 1 of 16 is 00 address 0x3ffc908c
[12:08:26][I][jk_rs485_bms:866]: [01] gps_heartbeat_switch_                is bit 2 of 16 is 00 address 0x3ffc91ec
[12:08:26][I][jk_rs485_bms:869]: [01] port_selection_switch_               is bit 3 of 16 is 00 address 0x3ffc9244
[12:08:26][I][jk_rs485_bms:874]: [01] disable_pcl_module_switch_           is bit 7 of 16 is 00 address 0x3ffc913c <------------
[12:08:26][I][jk_rs485_bms:877]: [01] display_always_on_switch_            is bit 4 of 16 is 01 address 0x3ffc8ea8
[12:08:26][I][jk_rs485_bms:880]: [01] special_charger_switch_              is bit 5 of 16 is 00 address 0x3ffc929c
[12:08:26][I][jk_rs485_bms:883]: [01] smart_sleep_on_switch_               is bit 6 of 16 is 00 address 0x3ffc90e4
[12:08:30][I][jk_rs485_bms:859]: [02] heating_switch_                      is bit 0 of 16 is 00 address 0x3ffc9588
[12:08:30][I][jk_rs485_bms:863]: [02] disable_temperature_sensors_switch_  is bit 1 of 16 is 00 address 0x3ffc8fb0
[12:08:30][I][jk_rs485_bms:866]: [02] gps_heartbeat_switch_                is bit 2 of 16 is 00 address 0x3ffc9844
[12:08:30][I][jk_rs485_bms:869]: [02] port_selection_switch_               is bit 3 of 16 is 00 address 0x3ffc989c
[12:08:30][I][jk_rs485_bms:874]: [02] disable_pcl_module_switch_           is bit 7 of 16 is 00 address 0x3ffc9794 <------------
[12:08:30][I][jk_rs485_bms:877]: [02] display_always_on_switch_            is bit 4 of 16 is 01 address 0x3ffc94d8
[12:08:30][I][jk_rs485_bms:880]: [02] special_charger_switch_              is bit 5 of 16 is 00 address 0x3ffc98f4
[12:08:30][I][jk_rs485_bms:883]: [02] smart_sleep_on_switch_               is bit 6 of 16 is 00 address 0x3ffc973c
[12:08:38][I][jk_rs485_bms:859]: [01] heating_switch_                      is bit 0 of 16 is 00 address 0x3ffc8f58
[12:08:38][I][jk_rs485_bms:863]: [01] disable_temperature_sensors_switch_  is bit 1 of 16 is 00 address 0x3ffc908c
[12:08:38][I][jk_rs485_bms:866]: [01] gps_heartbeat_switch_                is bit 2 of 16 is 00 address 0x3ffc91ec
[12:08:38][I][jk_rs485_bms:869]: [01] port_selection_switch_               is bit 3 of 16 is 00 address 0x3ffc9244
[12:08:38][I][jk_rs485_bms:874]: [01] disable_pcl_module_switch_           is bit 7 of 16 is 00 address 0x3ffc913c <------------
[12:08:38][I][jk_rs485_bms:877]: [01] display_always_on_switch_            is bit 4 of 16 is 01 address 0x3ffc8ea8
[12:08:38][I][jk_rs485_bms:880]: [01] special_charger_switch_              is bit 5 of 16 is 00 address 0x3ffc929c
[12:08:38][I][jk_rs485_bms:883]: [01] smart_sleep_on_switch_               is bit 6 of 16 is 00 address 0x3ffc90e4
[12:08:41][I][jk_rs485_bms:859]: [02] heating_switch_                      is bit 0 of 16 is 00 address 0x3ffc9588
[12:08:41][I][jk_rs485_bms:863]: [02] disable_temperature_sensors_switch_  is bit 1 of 16 is 00 address 0x3ffc8fb0
[12:08:41][I][jk_rs485_bms:866]: [02] gps_heartbeat_switch_                is bit 2 of 16 is 00 address 0x3ffc9844
[12:08:41][I][jk_rs485_bms:869]: [02] port_selection_switch_               is bit 3 of 16 is 00 address 0x3ffc989c
[12:08:41][I][jk_rs485_bms:874]: [02] disable_pcl_module_switch_           is bit 7 of 16 is 00 address 0x3ffc9794 <------------
[12:08:41][I][jk_rs485_bms:877]: [02] display_always_on_switch_            is bit 4 of 16 is 01 address 0x3ffc94d8
[12:08:41][I][jk_rs485_bms:880]: [02] special_charger_switch_              is bit 5 of 16 is 00 address 0x3ffc98f4
[12:08:41][I][jk_rs485_bms:883]: [02] smart_sleep_on_switch_               is bit 6 of 16 is 00 address 0x3ffc973c
[12:08:50][I][jk_rs485_bms:859]: [01] heating_switch_                      is bit 0 of 16 is 00 address 0x3ffc8f58
[12:08:50][I][jk_rs485_bms:863]: [01] disable_temperature_sensors_switch_  is bit 1 of 16 is 00 address 0x3ffc908c
[12:08:50][I][jk_rs485_bms:866]: [01] gps_heartbeat_switch_                is bit 2 of 16 is 00 address 0x3ffc91ec
[12:08:50][I][jk_rs485_bms:869]: [01] port_selection_switch_               is bit 3 of 16 is 00 address 0x3ffc9244
[12:08:50][I][jk_rs485_bms:874]: [01] disable_pcl_module_switch_           is bit 7 of 16 is 00 address 0x3ffc913c <------------
[12:08:50][I][jk_rs485_bms:877]: [01] display_always_on_switch_            is bit 4 of 16 is 01 address 0x3ffc8ea8
[12:08:50][I][jk_rs485_bms:880]: [01] special_charger_switch_              is bit 5 of 16 is 00 address 0x3ffc929c
[12:08:50][I][jk_rs485_bms:883]: [01] smart_sleep_on_switch_               is bit 6 of 16 is 00 address 0x3ffc90e4
[12:08:59][I][jk_rs485_bms:859]: [01] heating_switch_                      is bit 0 of 16 is 00 address 0x3ffc8f58
[12:08:59][I][jk_rs485_bms:863]: [01] disable_temperature_sensors_switch_  is bit 1 of 16 is 00 address 0x3ffc908c
[12:08:59][I][jk_rs485_bms:866]: [01] gps_heartbeat_switch_                is bit 2 of 16 is 00 address 0x3ffc91ec
[... cut ...]
[12:11:07][I][jk_rs485_bms:869]: [02] port_selection_switch_               is bit 3 of 16 is 00 address 0x3ffc989c
[12:11:07][I][jk_rs485_bms:874]: [02] disable_pcl_module_switch_           is bit 7 of 16 is 00 address 0x20336 <------------
[12:11:07][I][jk_rs485_bms:877]: [02] display_always_on_switch_            is bit 4 of 16 is 01 address 0x3ffc94d8
[12:11:07][I][jk_rs485_bms:880]: [02] special_charger_switch_              is bit 5 of 16 is 00 address 0x3ffc98f4
[12:11:07][I][jk_rs485_bms:883]: [02] smart_sleep_on_switch_               is bit 6 of 16 is 00 address 0x3ffc973c

```

txubelaxu, 2024-06-20:
I have made a lot of tests with standard ESP32. I will test with a ESP32-S3 (when they arrive), as told by @Sleeper85 
So... I need some days from now.

txubelaxu, 2024-06-22:
Tested with a ESP32-S3 and the issue still is there. :-(

txubelaxu, 2024-06-22:
..../.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/config/sdkconfig.h:68: warning:
-----> "CONFIG_ARDUINO_LOOP_STACK_SIZE"

cinusik, 2024-06-22:
Hope that this will be it .

syssi, 2024-06-23:
```
esphome:
  platformio_options:
    build_flags: 
      - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=17384
```

Instead of changing the code you could try to apply this define.

txubelaxu, 2024-06-23:
Thanks @syssi. I made that change yesterday, but the problem is still there.
It is happening only with switch_::Switch

Now is happening with port_selection_switch_:
```
[13:50:44][V][jk_rs485_sniffer:487]: free_heap 192.000000 kBytes [buffer: 120 bytes]
[13:50:44][V][jk_rs485_sniffer:487]: free_heap 192.000000 kBytes [buffer: 308 bytes]
[13:50:45][V][jk_rs485_sniffer:487]: free_heap 192.000000 kBytes [buffer: 120 bytes]
[13:50:45][V][jk_rs485_sniffer:487]: free_heap 192.000000 kBytes [buffer: 300 bytes]
[13:50:49][V][jk_rs485_sniffer:487]: free_heap 180.000000 kBytes [buffer: 420 bytes]
[13:50:49][V][jk_rs485_sniffer:487]: free_heap 164.000000 kBytes [buffer: 188 bytes]
[13:50:49][V][jk_rs485_sniffer:487]: free_heap 192.000000 kBytes [buffer: 428 bytes]
[13:50:49][V][jk_rs485_sniffer:487]: free_heap 192.000000 kBytes [buffer: 240 bytes]
[13:50:49][V][jk_rs485_sniffer:487]: free_heap 192.000000 kBytes [buffer: 308 bytes]
[13:50:50][V][jk_rs485_sniffer:487]: free_heap 192.000000 kBytes [buffer: 120 bytes]
[13:50:50][V][jk_rs485_sniffer:487]: free_heap 192.000000 kBytes [buffer: 308 bytes]
[13:50:50][V][jk_rs485_bms:1146]: Publishing state 1 for object with address 0x3fca1e74 [192.000000]
[13:50:50][V][jk_rs485_bms:1148]:   --------------------------------------- PUBLISHED     0x3FCA1E74 publish_state(state) of rs485-bms1 charging
[13:50:50][V][jk_rs485_bms:1146]: Publishing state 1 for object with address 0x3fca1f50 [192.000000]
[13:50:50][V][jk_rs485_bms:1148]:   --------------------------------------- PUBLISHED     0x3FCA1F50 publish_state(state) of rs485-bms1 discharging
[13:50:50][V][jk_rs485_bms:1146]: Publishing state 1 for object with address 0x3fca1fa8 [192.000000]
[13:50:50][V][jk_rs485_bms:1148]:   --------------------------------------- PUBLISHED     0x3FCA1FA8 publish_state(state) of rs485-bms1 balancing
[13:50:50][I][jk_rs485_bms:860]: [02] heating_switch_                      is bit 0 of 16 is 39 address 0x3fca1a28
[13:50:50][V][jk_rs485_bms:1146]: Publishing state 57 for object with address 0x3fca1a28 [192.000000]
[13:50:50][V][jk_rs485_bms:1148]:   --------------------------------------- PUBLISHED     0x3FCA1A28 publish_state(state) of rs485-bms1 heating
[13:50:50][I][jk_rs485_bms:864]: [02] disable_temperature_sensors_switch_  is bit 1 of 16 is 39 address 0x3fca220c
[13:50:50][V][jk_rs485_bms:1146]: Publishing state 57 for object with address 0x3fca220c [192.000000]
[13:50:50][V][jk_rs485_bms:1148]:   --------------------------------------- PUBLISHED     0x3FCA220C publish_state(state) of rs485-bms1 disable temperature sensors
[13:50:50][I][jk_rs485_bms:867]: [02] gps_heartbeat_switch_                is bit 2 of 16 is 39 address 0x3fca236c
[13:50:50][V][jk_rs485_bms:1146]: Publishing state 57 for object with address 0x3fca236c [192.000000]
[13:50:50][V][jk_rs485_bms:1148]:   --------------------------------------- PUBLISHED     0x3FCA236C publish_state(state) of rs485-bms1 gps heartbeat
[13:50:50][I][jk_rs485_bms:870]: [02] port_selection_switch_               is bit 3 of 16 is 39 address 0xff2d
[13:50:50][V][jk_rs485_bms:1150]:   --------------------------------------- NOT PUBLISHED 0xFF2D
[13:50:50][I][jk_rs485_bms:875]: [02] disable_pcl_module_switch_           is bit 7 of 16 is 39 address 0x3fca22bc <------------
[13:50:50][V][jk_rs485_bms:1146]: Publishing state 57 for object with address 0x3fca22bc [192.000000]
[13:50:50][V][jk_rs485_bms:1148]:   --------------------------------------- PUBLISHED     0x3FCA22BC publish_state(state) of rs485-bms1 disable pcl module
[13:50:50][I][jk_rs485_bms:878]: [02] display_always_on_switch_            is bit 4 of 16 is 39 address 0x3fca2000
[13:50:50][V][jk_rs485_bms:1146]: Publishing state 57 for object with address 0x3fca2000 [192.000000]
[13:50:50][V][jk_rs485_bms:1148]:   --------------------------------------- PUBLISHED     0x3FCA2000 publish_state(state) of rs485-bms1 display_always_on
[13:50:50][I][jk_rs485_bms:881]: [02] special_charger_switch_              is bit 5 of 16 is 39 address 0x3fca241c
[13:50:50][V][jk_rs485_bms:1146]: Publishing state 57 for object with address 0x3fca241c [192.000000]
[13:50:50][V][jk_rs485_bms:1148]:   --------------------------------------- PUBLISHED     0x3FCA241C publish_state(state) of rs485-bms1 special charger
[13:50:50][I][jk_rs485_bms:884]: [02] smart_sleep_on_switch_               is bit 6 of 16 is 39 address 0x3fca2264
[13:50:50][V][jk_rs485_bms:1146]: Publishing state 57 for object with address 0x3fca2264 [192.000000]
[13:50:50][V][jk_rs485_bms:1148]:   --------------------------------------- PUBLISHED     0x3FCA2264 publish_state(state) of rs485-bms1 smart sleep on
[13:50:50][V][jk_rs485_bms:1146]: Publishing state 0 for object with address 0x3fca2314 [192.000000]
[13:50:50][V][jk_rs485_bms:1148]:   --------------------------------------- PUBLISHED     0x3FCA2314 publish_state(state) of rs485-bms1 timed stored data
[13:50:50][V][jk_rs485_bms:1146]: Publishing state 1 for object with address 0x3fca21b4 [192.000000]
[13:50:50][V][jk_rs485_bms:1148]:   --------------------------------------- PUBLISHED     0x3FCA21B4 publish_state(state) of rs485-bms1 charging float mode
```

I have inserted a verify previous to the publish_state call to avoid the failure, but that is not the solution:
```
void JkRS485Bms::publish_state_(switch_::Switch *obj, const bool &state) {
  if (obj == nullptr) {
    ESP_LOGE(TAG, "Object is nullptr");
    return;
  }
[... cut ...]
  if (reinterpret_cast<uintptr_t>(obj) > 0x3f000000) {
    ESP_LOGV(TAG, "Publishing state %d for object with address %p [%f]", state, (void*)obj, ((float)free_heap/1024));
    obj->publish_state(state);
    ESP_LOGV(TAG, "  --------------------------------------- PUBLISHED     0x%02X publish_state(state) of %s", reinterpret_cast<uintptr_t>(obj), obj->get_name().c_str());
  } else {
    ESP_LOGV(TAG, "  --------------------------------------- NOT PUBLISHED 0x%02X", reinterpret_cast<uintptr_t>(obj));
  }

}
```

txubelaxu, 2024-06-25:
I think the bug has been corrected. I have uploaded the code just now.
I think the code should work with no problem on both: esp32 or esp32-S3. It was not a memory problem. It was a JkRS485BmsSwitch problem.

To verify...

txubelaxu, 2024-06-25:
After last change, I see that esphome-jk-bms-can restarts periodically. 
But I think it is because a different reason. So there were 2 and 1 has been solved (I think).

cinusik, 2024-06-25:
@txubelaxu Testing, testing... So far, so good ;) I had one restart after 12 minutes so I tried to change two parameters from:
```
bms_update_interval: '3s' # going below '3s' can cause problems
bms_combine_interval: '1s'
```
to
```
bms_update_interval: '5s' # going below '3s' can cause problems
bms_combine_interval: '2s'
```
and now 1hour and 30 minuter without single issue.

txubelaxu, 2024-06-25:
Hi @cinusik. With those settings here is restarting after 1 minute.
I have 2 BMSs.

cinusik, 2024-06-25:
Hi @txubelaxu,
I also have 2 BMSes and AtomS3. Over 7 hours without problems thanks to your fix.
No idea why this is not working for you?
I had restarts around 1 minute or less when I did not have ground wire connection. When I connected GND from JK BMS to RS485 converter than ESP became more stable working randomly like around 10 minutes, half hour, one our etc - until todays fix. Now is perfectly stable.
Try to connect ground wire if you have not connected it yet.

txubelaxu, 2024-06-26:
Testing new improvements and bug corrected

cinusik, 2024-06-26:
> Testing new improvements and bug corrected

Testing it also and what I fond is that switches do not work. When I turn off charging using bluetooth phone app the change is not reflected in HA

txubelaxu, 2024-06-26:
ups, sorry @cinusik. I changed settings_info (frame_type=1) to a frequency of 1/1h. 
I have changed to 1/5s

Could you test the code I have just uploaded, please?

cinusik, 2024-06-26:
@txubelaxu Confirmed, works as expected 🥇 
Do you plan to make switches "life" instead of read only? like BLE version?

txubelaxu, 2024-06-26:
Yes, that's the plan. I'll try.

Many thanks!!!

txubelaxu, 2024-06-28:
OK, confirmed!! I have esphome-jk-bms-can working for 
[image]

Everything is showing quite stable.

[26 more replies]

closed 2024-08-10 by txubelaxu