Sleeper85/esphome-jk-bms-can #56: Instruction=Status=Bulk

txubelaxu, 2024-07-03:
In my opinion, when **Instruction** is **Bulk**, but the **I=0A**, then the status should be "**Idle**" (or something like that)

[image]

Sleeper85, 2024-07-03:
Hi,

I have seen your messages as well as the PR. I'm currently busy with another project. I'll look into it and get back to you soon.

shvmm, 2024-07-03:
Current being 0 while in **Bulk** represents a specific situation/mitigation in the charging logic loop.
What were the exact conditions that caused this to happen?

Adding a status named Idle is easy but the charging logic depends on the specific states to communicate values to the inverter.

txubelaxu, 2024-07-03:
OK. I undestand.

The conditions were: 2xBMS in a stationary status and no inverter conected
to the system.

Many thanks for the info!!

Sleeper85, 2024-07-12:
As @shvmm says the only way to exit a Bulk phase is for it to end with a fully charged battery, no matter if the current is zero, negative or positive.

closed 2024-07-12 by Sleeper85