Sleeper85/esphome-jk-bms-can #58: Float charging not respected

cinusik, 2024-07-09:
For the last few days I'm testing RS485 version fixed by @txubelaxu 
It is stable now and working almost perfect.
I observed that float charging is not respected. After Bulk phase charging is finished and it is going from Balancing/Absorption state to Floating state. But it not stays on 53.6V but it is going down to like 53.3 or 53.2.
I attached pictures from my Solar Assistant chart from today, it was full sun all day.
Is this bug?
[image]

Sleeper85, 2024-07-09:
The 4 sensors below show what is requested from the inverter.
[My experience with Deye is that he doesn't always do what is requested of him.](https://diysolarforum.com/threads/deye-weird-voltage-behavior-when-time-of-use-is-enabled.80908/)

You must therefore analyze what was requested of your Growatt inverter and check whether it has complied with the instructions.

[image]

cinusik, 2024-07-10:
Thank you for the link, interesting lecture about Deye. Will try to investigate it more and will observe float voltage request in Solar assistant on inverter settings page what requested there.

cinusik, 2024-07-11:
Looks like requested float voltage is passed to the inverter correctly. It's not an issue with SmartBMS/YamBMS - it's the way how inverter behave itself. Looks like inverter is trying to float at 53.6V but if there is some load at float time and there is not enough sun than voltage drop for example to 53.4V, than after load is gone inverter is trying to float at that 53.4 voltage. It is not trying to go to 53.6V - it is just floating at battery voltage at this particular moment - probably until rebulk voltage setting. That's what I observe and assume :)

closed 2024-07-24 by Sleeper85