Sleeper85/esphome-jk-bms-can #61: Compiled OK but not working for me this last development version (yambms)

txubelaxu, 2024-08-05:
I have tested the esphome-jk-bms last version with my config and it goes OK.
But when I launch esphome-jk-bms-can last version (development branch), there is a not normal behaviour comparing to previous version (previous yambms). I have tried to tune the yamls to be similar to those that I have working, but I cannot get it working.

I am testing more to know why it is happening to my system.

Sleeper85, 2024-08-05:
What problem are you facing exactly ?

txubelaxu, 2024-08-05:
I am running multi-bms_JK-PB_RS485_mode2.yaml

It starts receiving some info, but after some seconds, it stops:

[12:56:11][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x0D
[0010000000000000]
[12:56:12][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x0E
[0010000000000000]
[12:56:13][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x0F
[0010000000000000]
[12:56:14][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x01
[0010000000000000]
[12:56:15][I][jk_rs485_sniffer:346]: SILENCE: 1157.000000 ms
[12:56:15][D][jk_rs485_sniffer:365]:
..........................................
[12:56:15][D][jk_rs485_sniffer:369]: Buffer before number 1:
 55.AA.EB.90.02.05.F2.0C.F1.0C.F1.0C.F1.0C.F2.0C.F1.0C.F1.0C.F1.0C.F2.0C.F2.0C.F2.0C.F2.0C.F2.0C.F2.0C.F2.0C.F0.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.F2.0C.01.00.00.01.53.00.50.00.4F.00.4A.00.4D.00.4B.00.4D.00.4D.00.53.00.4E.00.4D.00.4A.00.4C.00.4D.00.52.00.51.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.0A.01.00.00.00.00.1C.CF.00.00.00.00.00.00.00.00.00.00.F0.00.F6.00.00.00.00.00.00.00.00.61.08.7E.04.00.80.A3.04.00.00.00.00.00.23.8F.00.00.64.00.00.00.68.8F.1E.01.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.9A.03.00.00.00.00.60.54.3F.40.00.00.00.00.B6.14.00.00.00.01.01.01.00.06.00.00.38.B0.ED.09.00.00.00.00.0A.01.EF.00.F9.00.9A.03.FF.DD.A4.08.9C.00.00.00.80.51.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DD.2F.01.01.B0.07.00.00.00.9D.01.10.16.20.00.01.04.4B
(308)
[12:56:15][D][jk_rs485_sniffer:640]:
###############################Sequence found SIZE: 308
[12:56:15][I][jk_rs485_sniffer:512]: NODES AVAILABLE CHANGED: address 0x01
(0->1) [0010000000000000] --> [0110000000000000]
[12:56:15][D][jk_rs485_sniffer:699]: Frame received from SLAVE (type: 0x02,
309 bytes) 01 address
[12:56:16][D][jk_rs485_sniffer:074]: Received Event from BMS..
[address:0x01] @ 2 -->  WORKING ! #####
[12:56:16][D][jk_rs485_sniffer:084]: updated
last_cell_info_request_received_OK
[12:56:16][D][jk_rs485_sniffer:371]: Response:            12:
[12:56:16][D][jk_rs485_sniffer:381]: Buffer empty
[12:56:16][W][component:237]: Component jk_rs485_sniffer took a long time
for an operation (334 ms).
[12:56:16][C][jk_rs485_sniffer:722]: JkRS485Sniffer:
[12:56:16][C][jk_rs485_sniffer:723]:   RX timeout: 500 ms
[12:56:17][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x03
[0110000000000000]
[12:56:18][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x04
[0110000000000000]
[12:56:19][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x05
[0110000000000000]
[12:56:20][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x06
[0110000000000000]
[12:56:21][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x07
[0110000000000000]
[12:56:22][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x08
[0110000000000000]
[12:56:23][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x09
[0110000000000000]
[12:56:24][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x0A
[0110000000000000]
[12:56:25][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x0B
[0110000000000000]
[12:56:26][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x0C
[0110000000000000]
[12:56:27][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x0D
[0110000000000000]
[12:56:28][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x0E
[0110000000000000]
[12:56:36][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x08
[0110000000000000]
[12:56:37][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x09
[0110000000000000]
[12:56:38][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x0A
[0110000000000000]
[12:56:39][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x0B
[0110000000000000]
[12:56:40][D][jk_rs485_sniffer:417]: SCANNING TO DISCOVER...0x0C
[0110000000000000]

I have checked that ESP32 is sending to network from another RS485 client.

txubelaxu, 2024-08-05:
It is very very incredible: when I touch some free pins of the esp32 with
my hands, it starts working!

Astonished!

lol

Sleeper85, 2024-08-05:
> It is very very incredible: when I touch some free pins of the esp32 with my hands, it starts working!

Amazing! are all GND pins connected together? Including to the RS485 bus ?

txubelaxu, 2024-08-05:
Yes, of course. I thought about a grounding problem, as well. But it works
great with esphome-jk-bms...

It reverse to the same issue after some seconds, and If I touch some pins
it recovers. I think it is related more to the can-pins.
I am not using UART to CAN converter. Could I somehow disable this
"can-layer"?

txubelaxu, 2024-08-05:
Anyway, I have just tested again the previous esphome-jk-bms-can version,
and it works great with the same hardware configuration (I have changed
nothing), and I don't need to touch the pins to recover...

Sleeper85, 2024-08-05:
> Anyway, I have just tested again the previous esphome-jk-bms-can version,
and it works great with the same hardware configuration (I have changed
nothing), and I don't need to touch the pins to recover...

Which version? Old 1.17.5 or YamBMS 1.4.1?

Sleeper85, 2024-08-05:
[Last change since the version without TALK PIN.](https://diysolarforum.com/threads/yambms-jk-bms-can-with-new-cut-off-charging-logic-open-source.79325/page-63#post-1168509)

Sleeper85, 2024-08-05:
There have been major changes in the `yambms_canbus.yaml` file since `YamBMS 1.4.2`.

See the [changelog](https://github.com/Sleeper85/esphome-jk-bms-can/tree/development?tab=readme-ov-file#changelog) :

* CANBUS 2.3.1 : Improved the procedure for sending canbus frames with reduced loop time, rewritten of the canbus link validation code and added `Inverter Heartbeat Monitoring` function
* CANBUS 2.3.2 : Added `LuxPower` protocol with updated `can_id` 0x355, 0x356, 0x359 and 0x35C

You can remove the import of the `canbus1` package to test.

```YAML
  canbus1: !include
    file: packages/yambms/yambms_canbus.yaml
    vars:
      # YamBMS ID
      yambms_id: 'yambms1'
      # CANBUS vars
      canbus_id: 'canbus1'
      canbus_name: 'CANBUS 1'
      canbus_node_id: 'canbus_node1'
      canbus_light_id: 'esp_light'
      # The CANBUS link will be considered down if no response from the inverter (ID 0x305) for 5s
      # The Deye inverter sends the ACK (can_id 0x305) only when it receives the can_id 0x356
      canbus_link_timer: '5'
```

Sleeper85, 2024-08-05:
@virus100b just shared this with us on the DIY solar forum.

> No issues since i have replace can board.
I forget to mention that i have replace the talk pin RS485 board with the one without talk pin(The one with LEDs on TX, RX and voltage supressor on A, B ) with no issues. Today i notice the @Txubelaxu modified the code to work without talk pin but i did not update yet.
For now is working fine. Uptime for over 24h.

He is now using a RS485 board without TALK PIN.
I asked him to come here to talk about it.

Sleeper85, 2024-08-05:
@txubelaxu 

Not using canbus with your inverter ?

txubelaxu, 2024-08-05:
I have not one yet. :-)

I am creating my system from bottom to top. lol

Sleeper85, 2024-08-05:
Maybe the problem is that you don't have a `canbus` board ?

More tests should be done now to confirm that the latest `YamBMS 1.4.2` and `CANBUS 2.3.2` code still works correctly with your RS485 component.

There are currently 2 guys using your RS485 component with my project @widget4145 and @virus100b

A 3rd guy will soon be using your RS485 component with my project and the isolated RS485 board without TALK pin.

@widget4145 has problems and can't monitor more than 5 BMS with his Atom S3 and this also happens when he uses your code only without YamBMS.
Did this happen after upgrading your component to version `1.3` or `1.3.1` ?
Or is it related to the version of `esphome` or `HA`?
He recompiled again yesterday with your latest changes but it still doesn't work with his 7 BMS.

txubelaxu, 2024-08-05:
Yes, it was the problem. Removing can component from yaml, it works (YamBMS 1.4.2)

I have analyzed @widget4145 user's issue about why is not working with 7 BMSs. I cannot recreate that config here.
From 1.3 to 1.3.1 there is a bug correction on the code:

[image]

That is why I unpupblished v1.3.

Sleeper85, 2024-08-05:
@widget4145 has been using your component for a long time and in early May 2024 he was the first to use my multi-bms version with your component in version `1.2`. Everything worked well for 2 months with 7x BMS and an Atom S3.

Then you fixed bugs and released version `1.3.1`.

I suggested to @widget4145 to recompile the `same` multi-bms code that I had provided to him to benefit from your bug fixes and since then he has never managed to monitor more than 5 BMS. At that time he also updated `esphome` and `HA`.

Do you have a backup of version `1.2` of your components ?

We could thus check if the problem comes from the latest changes or not ?

I still have the YAML that I sent him in May and June when your component was in version `1.2`.

txubelaxu, 2024-08-05:
I will review the changes:
https://github.com/txubelaxu/esphome-jk-bms/compare/v1.2...v1.3.1

txubelaxu, 2024-08-05:
> Yes, it was the problem. Removing can component from yaml, it works (YamBMS 1.4.2)
>
> I have analyzed @widget4145 user's issue about why is not working with 7 BMSs. I cannot recreate that config here. [...]

Well... it works for some time. After a few minutes it restarts. :-(

txubelaxu, 2024-08-06:
I have checked esphome-jk-bms and has worked for all the night with no reboot:
[image]

But something happens to esphome-jk-bms-can

virus100b, 2024-08-06:
For me no issues so far. Using yambms 1.4.2 version with  RS485 board without talk pin. i am using previous txubelaxu component. I did not update to last verdion.

[image]

txubelaxu, 2024-08-06:
Maybe it is because I deactivated the can layer. But... without can layer,
should it work as stable as esphome-jk-bms?

Sleeper85, 2024-08-06:
> Maybe it is because I deactivated the can layer. But... without can layer, should it work as stable as esphome-jk-bms?

Normally yes. The CAN layer only retrieves information from YamBMS and sends it over the CAN bus.
The canbus is declared in `board.yaml` but I don't think it should cause reboot problems even if you don't use it.

Which ESP32 board do you use?
How many BMS?

Uptime of my ESP32 DevKit V1 with YamBMS 1.4.2 and 2x BMS (UART + BLE).
The BLE stack takes a lot of memory, I'm at the limit but it's stable.

[image]

Same ESP32 with only 1x BMS UART, noticed the difference in available memory without the BLE stack.

[image]

Did you compile the latest version from my git or from your fork (it's not up to date) ?

txubelaxu, 2024-08-06:
I have 2xJK-PB connected using RS485. Addresses: 0x01 and 0x02. So, I am
using mode2.

I got the "development" branch from your git yesterday. I have syncrhonized
again now and I have compiled again with the this last version.
After this same issue appears.

I think the esp32 is very powerful for this code:

Successfully created esp32s3 image.
Linking .pioenvs/esp32-s3/firmware.elf
RAM:   [=         ]   6.6% (used 34624 bytes from 524288 bytes)
Flash: [=         ]  10.9% (used 881893 bytes from 8126464 bytes)
Building .pioenvs/esp32-s3/firmware.bin
Creating esp32s3 image...
Successfully created esp32s3 image.
esp32_create_combined_bin([".pioenvs/esp32-s3/firmware.bin"],
[".pioenvs/esp32-s3/firmware.elf"])

The issue: after some seconds/minutes esp32 restarts

txubelaxu, 2024-08-06:
Hi @Sleeper85,

I have just made a change on my git. Could @widget4145 try with this change?

txubelaxu, 2024-08-06:
I have opened an independent issue for "7 slaves issue".

I have detected that 

esphome-jk-bms:
```
esphome logs esp32-S3-example-jkpb-rs485_0master_2slaves.yaml | grep "nullptr\|for object with address"
INFO ESPHome 2024.7.3
INFO Reading configuration esp32-S3-example-jkpb-rs485_0master_2slaves.yaml...
INFO Starting log output from esp-rs485-gw.local using esphome API
INFO Successfully connected to esp-rs485-gw @ 172.26.1.217 in 0.089s
INFO Successful handshake with esp-rs485-gw @ 172.26.1.217 in 0.523s
[20:36:43][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca30d0 [184.000000]
[20:36:45][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca2ad0 [184.000000]
[20:36:45][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca2b28 [184.000000]
[20:36:45][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca2c04 [184.000000]
[20:36:46][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca2c5c [184.000000]
[20:36:46][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca2cb4 [184.000000]
[20:36:46][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca2684 [184.000000]
[20:36:46][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca2e10 [184.000000]
[20:36:46][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca2e68 [184.000000]
[20:36:46][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca2ec0 [184.000000]
[20:36:46][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca2f18 [184.000000]
[20:36:46][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca2f70 [184.000000]
[20:36:46][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca2fc8 [184.000000]
[20:36:46][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca3020 [184.000000]
[20:36:48][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca2524 [184.000000]
[20:36:48][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca257c [184.000000]
[20:36:48][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca25d4 [184.000000]
[20:36:48][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca262c [184.000000]
[20:36:48][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca2708 [184.000000]
[20:36:48][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca2760 [184.000000]
[20:36:48][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca27b8 [184.000000]
[20:36:48][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca2810 [184.000000]
[20:36:48][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca2868 [184.000000]
[20:36:48][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca28c0 [184.000000]
[20:36:48][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca2918 [184.000000]
[20:36:48][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca2970 [184.000000]
[20:36:48][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca29c8 [184.000000]

```

but, esphome-jk-bms-can:
esphome logs multi-bms_JK-PB_RS485_mode2.yaml | grep "nullptr\|for object with address"
INFO ESPHome 2024.7.3
INFO Reading configuration multi-bms_JK-PB_RS485_mode2.yaml...
INFO Detected timezone 'Europe/Madrid'
INFO Starting log output from esp32-s3.local using esphome API
INFO Successfully connected to esp32-s3 @ 172.26.1.217 in 0.488s
INFO Successful handshake with esp32-s3 @ 172.26.1.217 in 0.022s
```
[20:25:45][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca8770 [8064.000000]
[20:25:46][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca87c8 [8064.000000]
[20:25:46][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca8820 [8064.000000]
[20:25:46][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca8878 [8064.000000]
[20:25:46][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca88d0 [8064.000000]
[20:25:47][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca8928 [8064.000000]
[20:25:47][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca8980 [8064.000000]
[20:25:47][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca89d8 [8064.000000]
[20:25:47][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca8a30 [8064.000000]
[20:25:47][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca8484 [8064.000000]
[20:25:47][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca8b8c [8064.000000]
[20:25:47][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca8be4 [8064.000000]
[20:25:47][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca8c3c [8064.000000]
[20:25:47][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca8c94 [8064.000000]
[20:25:49][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca5200 [8064.000000]
[20:25:49][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca8274 [8064.000000]
[20:25:49][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca82cc [8064.000000]
[... cut ...]
[20:25:49][V][jk_rs485_bms:1292]: Publishing state 1 for object with address 0x3fca86c0 [8064.000000]
[20:25:50][V][jk_rs485_bms:1292]: Publishing state 0 for object with address 0x3fca8d44 [8064.000000]

```

This is an issue that I corrected, about a corruption on certain zone of memory caused because when creating instances without arguments, it is necessary a "default constructor" (that is why I added line 17):

[image]

So, the problem reapears again here, but only with "esphome-jk-bms-can" and not with "esphome-jk-bms" alone. 
Umm... I have think a lot about this.
lololololol

Any ideas?

Sleeper85, 2024-08-06:
@txubelaxu 

ESP32-S3 is powerful enough for sure.

I just compiled the code on an ESP32-S3 without importing the `yambms_canbus.yaml` package with supervision of 3x JK-B* BMS.
We'll see if I encounter the same problem as you or not.

Wait and see

Sleeper85, 2024-08-07:
@txubelaxu 

[image]

Here is the uptime of my `ESP32-S3` test board without the `yambms_canbus.yaml` package, so no reboot at home.

The only difference is that I'm not using your RS485 component because I do not have a JK-PB BMS.

But @virus100b who uses your RS485 component confirms that it is stable for him too.

If you think it comes from the canbus pins you could try to modify the `board_esp32-s3-devkitc-1.yaml` file and comment out the block below.

```YAML
canbus:
  # CANBUS NODE 1
  - platform: esp32_can
    id: canbus_node1
    tx_pin: 38
    rx_pin: 39
    can_id: 1
    bit_rate: 500kbps
```

If you don't have an inverter, you can easily simulate an inverter with two `CAN transceiver` boards I can explain how to do it if you are interested.

[SN65HVD230 3V3 does not require a 4.7K resistor.](https://github.com/Sleeper85/esphome-jk-bms-can/blob/development/documents/README/Supported_devices.md#supported-can-bus-transceiver)

virus100b, 2024-08-07:
I have not updated to the last @txubelaxu component. Below is my uptime using clasic esp32. 

[image]

Can module TJA1050 is hiden under esp32 board.

[image]

txubelaxu, 2024-08-07:
> @txubelaxu [...]

Yes, I have that code block commented out all the time.
After last changes I made last night, the "pointer bug" has gone away, but not the restarts. Please, do not ask why these changes have corrected that bug because I do not know. lol In the similar previous bug, I was googling and openaing and encountered that suggestion.

These are the rebootings I have detected during this night:
```
WARNING esp32-s3 @ 172.26.1.217: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-s3 @ 172.26.1.217
WARNING Disconnected from API
INFO Successfully connected to esp32-s3 @ 172.26.1.217 in 1.036s
INFO Successful handshake with esp32-s3 @ 172.26.1.217 in 0.022s
WARNING esp32-s3 @ 172.26.1.217: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-s3 @ 172.26.1.217
WARNING Disconnected from API
INFO Successfully connected to esp32-s3 @ 172.26.1.217 in 1.491s
INFO Successful handshake with esp32-s3 @ 172.26.1.217 in 0.018s
WARNING esp32-s3 @ 172.26.1.217: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-s3 @ 172.26.1.217
WARNING Disconnected from API
INFO Successfully connected to esp32-s3 @ 172.26.1.217 in 0.574s
INFO Successful handshake with esp32-s3 @ 172.26.1.217 in 0.021s
WARNING esp32-s3 @ 172.26.1.217: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-s3 @ 172.26.1.217
WARNING Disconnected from API
INFO Successfully connected to esp32-s3 @ 172.26.1.217 in 0.842s
INFO Successful handshake with esp32-s3 @ 172.26.1.217 in 0.018s
WARNING esp32-s3 @ 172.26.1.217: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-s3 @ 172.26.1.217
WARNING Disconnected from API
INFO Successfully connected to esp32-s3 @ 172.26.1.217 in 2.348s
INFO Successful handshake with esp32-s3 @ 172.26.1.217 in 0.024s
WARNING esp32-s3 @ 172.26.1.217: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-s3 @ 172.26.1.217
WARNING Disconnected from API
INFO Successfully connected to esp32-s3 @ 172.26.1.217 in 0.135s
INFO Successful handshake with esp32-s3 @ 172.26.1.217 in 0.115s
WARNING esp32-s3 @ 172.26.1.217: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-s3 @ 172.26.1.217
WARNING Disconnected from API
INFO Successfully connected to esp32-s3 @ 172.26.1.217 in 0.408s
INFO Successful handshake with esp32-s3 @ 172.26.1.217 in 0.021s
WARNING esp32-s3 @ 172.26.1.217: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-s3 @ 172.26.1.217
WARNING Disconnected from API
INFO Successfully connected to esp32-s3 @ 172.26.1.217 in 0.867s
INFO Successful handshake with esp32-s3 @ 172.26.1.217 in 0.020s
WARNING esp32-s3 @ 172.26.1.217: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-s3 @ 172.26.1.217
WARNING Disconnected from API
```

I will continue investigating this issue. 

Many thanks!!!

Sleeper85, 2024-08-07:
@txubelaxu 

WiFi problem ?

txubelaxu, 2024-08-07:
It could be, not sure 100%. It is not restarting when using "esphome-jk-bms". Only with "esphome-jk-bms-can". I have to test much more with settings.

txubelaxu, 2024-08-09:
I think I can sat that the problem has gone. 😃 

I think it was because Cell info and Settings were 1 time / 5s. I set the period of second one ("settings") to 10000ms and problem disapeared.

closed 2024-08-10 by txubelaxu