Sleeper85/esphome-jk-bms-can #63: 7 slaves issue

txubelaxu, 2024-08-06:
Hi @Sleeper85,

I have just made a change on my git. Could @widget4145 try with this change?

_Originally posted by @txubelaxu in https://github.com/Sleeper85/esphome-jk-bms-can/issues/61#issuecomment-2271587004_

Sleeper85, 2024-08-06:
Hi,

I will ask @widget4145 to recompile with the latest changes.

I will keep you informed.

Thanks

widget4145-1, 2024-08-10:
Hi, sorry, I forgot to reply here. I'd like to appreciate Sleeper85's persistence to help me. Thank you very much!

Unfortunately, with every recent changes have been made, it is still behaving the same. Anything more than 5 batteries freezes the values, but not feeezing the esp device per se.

I'm using Sleeper85's latest code by the way, deleted the device and clean build:
yambms 1.4.2 canbus 2.3.3

Here are the logs, please press to reveal the code:

<details><summary>Warning: Long Code</summary>
<p>

```

INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/multi-bms_JK-PB_RS485_mode2.yaml...
INFO Updating https://github.com/txubelaxu/esphome-jk-bms.git@main
WARNING GPIO35 is used by the PSRAM interface on ESP32-S3R8 / ESP32-S3R8V and should be avoided on these models
INFO Detected timezone 'Asia/Shanghai'
Failed config

sensor.jk_rs485_bms: [source /config/esphome/packages/bms/bms_JK-PB_RS485_bms_minimal.yaml:70]
  platform: jk_rs485_bms
  jk_rs485_bms_id: bms1
  balancing_direction: 
    name: yambms JK-BMS 1 balancing direction
  
  [cell_smart_sleep_voltage] is an invalid option for [sensor.jk_rs485_bms]. Did you mean [cell_average_voltage], [smart_sleep_time], [cell_delta_voltage]?
  cell_smart_sleep_voltage: 
    name: yambms JK-BMS 1 cell smart sleep voltage
  cell_voltage_min: 
    id: bms1_min_cell_voltage
    name: yambms JK-BMS 1 cell voltage min
  cell_voltage_max: 
    id: bms1_max_cell_voltage
    name: yambms JK-BMS 1 cell voltage max
  cell_voltage_min_cell_number: 
    id: bms1_min_voltage_cell
    name: yambms JK-BMS 1 cell voltage min cell number
  cell_voltage_max_cell_number: 
    id: bms1_max_voltage_cell
    name: yambms JK-BMS 1 cell voltage max cell number
  cell_delta_voltage: 
    name: yambms JK-BMS 1 cell delta voltage
  cell_average_voltage: 
    name: yambms JK-BMS 1 cell average voltage
  cell_resistance_min: 
    name: yambms JK-BMS 1 cell resistance min
  cell_resistance_max: 
    name: yambms JK-BMS 1 cell resistance max
  cell_resistance_min_cell_number: 
    name: yambms JK-BMS 1 cell resistance min cell number
  cell_resistance_max_cell_number: 
    name: yambms JK-BMS 1 cell resistance max cell number
  battery_voltage: 
    id: bms1_total_voltage
    name: yambms JK-BMS 1 battery voltage
  battery_current: 
    id: bms1_current
    name: yambms JK-BMS 1 battery current
  battery_power: 
    id: bms1_power
    name: yambms JK-BMS 1 battery power
  battery_power_charging: 
    id: bms1_charging_power
    name: yambms JK-BMS 1 battery power charging
  battery_power_discharging: 
    id: bms1_discharging_power
    name: yambms JK-BMS 1 battery power discharging
  temperature_sensor_1: 
[... 74 lines cut ...]
  [cell_soc0_voltage] is an invalid option for [sensor.jk_rs485_bms]. Did you mean [cell_voltage_24], [cell_voltage_23], [cell_voltage_22]?
  cell_soc0_voltage: 
    name: yambms JK-BMS 1 cell soc0 voltage
  
  [cell_undervoltage_protection] is an invalid option for [sensor.jk_rs485_bms]. Did you mean [cell_undervoltage_protection_release_time], [cell_overvoltage_protection_release_time], [cell_voltage_min]?
  cell_undervoltage_protection: 
    name: yambms JK-BMS 1 cell undervoltage protection
  
  [cell_undervoltage_protection_recovery] is an invalid option for [sensor.jk_rs485_bms]. Did you mean [cell_undervoltage_protection_release_time], [cell_overvoltage_protection_release_time], [powertube_temperature_protection_recovery]?
  cell_undervoltage_protection_recovery: 
    id: bms1_cell_uvpr
    name: yambms JK-BMS 1 cell undervoltage protection recovery
  
  [cell_power_off_voltage] is an invalid option for [sensor.jk_rs485_bms]. Did you mean [cell_average_voltage], [cell_delta_voltage], [cell_voltage_24]?
  cell_power_off_voltage: 
    name: yambms JK-BMS 1 cell power off voltage
  
  [max_charging_current] is an invalid option for [sensor.jk_rs485_bms]. Did you mean [max_discharging_current], [balancing_current], [charging_cycles]?
  max_charging_current: 
    id: bms1_max_charge_current
    name: yambms JK-BMS 1 max charging current

```
</p>
</details>

Sleeper85, 2024-08-10:
@txubelaxu 

I see a lot of `is an invalid option for` errors. What is it?

txubelaxu, 2024-08-10:
I moved from sensor onto number a lot of params. This way it is possible to
change any param of the BMS from the frontend.

I changed those yamls (full version) as well to avois those errors and
dashboards.

txubelaxu, 2024-08-10:
Hi @widget4145-1,

Could you try using the files in my git (https://github.com/txubelaxu/esphome-jk-bms-can) 
--> branch (development_txubelaxu) (https://github.com/txubelaxu/esphome-jk-bms-can/tree/development_txubelaxu)

I moved many params from sensor (read only) to number (read and write).
Broadcast mode is enabled by default, so if you need to change 1 param to 7 slaves, one change on one BMS is enough to send to the other 6 automatically.

Sleeper85, 2024-08-10:
@txubelaxu

@widget4145-1 uses a slightly modified version with a `yambms_custom.yaml`.

I'll go get the new YAML from your git and replace it in my current version.
Then I'll send the new version to @widget4145-1 so he can test.

widget4145-1, 2024-08-10:
Hi,

Thank you Sleeper85 for modifying txubelaxu's code according to my personal customization.

It would be rather challenging for me if I had done it by myself.

I'd like to let you know that the recent changes has been behaving as it should be now.

Thanks to both of you. And thanks to Sleeper85's persistence on solving this issue. He shouldn't have to, but he has been doing it on my behalf. Thank you txubelaxu as well for helping us, I'm really grateful.

I'll observe more and report back in case any issues arise.

txubelaxu, 2024-08-10:
Congratulations!!!

closed 2024-08-10 by txubelaxu