Sleeper85/esphome-jk-bms-can #68: RCV, RFV, RCV Time and RFV Time question

txubelaxu, 2024-08-13:
Hi everybody,

I am little confused about how RCV, RFV, RCV Time and RFV Time parameter CELL values are managed by esphome-jk-bms-can:

[image]

Does yambms respect (RCV * cell number) & (RFV * cell_number)?
After reaching the voltage of (RCV * cell number), does it get to maintain that voltage for "RCV Time"?
After reaching RCV Time, does it ask to the inverter for changing to (RFV * cell_number)?
After reaching the voltage of (RFV * cell_number), does it get to maintain that voltage for "RFV Time"?

I have seen that, for example, yambms RFV is changeable from the frontend.

Many thanks to everyone for your great work!!!

Sleeper85, 2024-08-13:
YamBMS was created to work with several different brands of BMS that do not have `RCV, RFV, RCV Time and RFV Time` parameters like in JK-PB BMS.

So YamBMS uses [its own charging logic](https://github.com/Sleeper85/esphome-jk-bms-can/blob/development/documents/README/Charging_logic.md) which is much more advanced than what `JK-PB` BMS does.

Regarding the charging values ​​which are non-existent in `JK-B` BMS, YamBMS uses its own values ​​modifiable in HA but also uses certain values ​​coming from the BMS configuration see [this Excel table](https://docs.google.com/spreadsheets/d/16Sa4z38WJtGPCSkVRymQaLzyHjK8hKauhE86SetRiB0/edit?usp=sharing).

I still have a lot of work to do to document all the functions.

[image]

[image]

shvmm, 2024-08-16:
> Hi everybody,
> 
> I am little confused about how RCV, RFV, RCV Time and RFV Time parameter CELL values are managed by esphome-jk-bms-can:

Highly likely these feature requests were made to the JK-BMS team by people unaware that cutoff charging logic exists.
People don't use a 16S inverter BMS to charge Lead Acid batteries. Vice-versa, fitting Lead Acid Charging behaviour to work for Lithium Cells is waste of time in the age of BMS communication.

I would much rather want JK_BMS team give us firmware update to turn balancing ON/OFF using UART on all their BMS.

closed 2024-09-02 by Sleeper85