Sleeper85/esphome-jk-bms-can #69: Install failed

dmsims, 2024-08-20:
Windows 11, Atom S3 lite

Prerequisites installed (all today) - Python, ESPhome, Git

Entered command:

David@DavidPC MINGW64 ~/esphome-jk-bms-can (main)
$ esphome run ESP32_LFP_Wire_jk-bms-can.yaml
INFO ESPHome 2024.7.3

Error:
Failed config
At least one platform must be specified for 'ota'; add 'platform: esphome' for original OTA functionality

Edited ESP32_LFP_Wire_jk-bms-can.yaml

added platform: esphome after line 167 (ota:)

Re ran install command and it proceeded until:

Tool Manager: Installing platformio/tool-idf @ ~1.0.1
INFO Installing platformio/tool-idf @ ~1.0.1
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-idf@1.0.1 has been installed!
INFO tool-idf@1.0.1 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40700.0
INFO Installing platformio/tool-scons @ ~4.40700.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40700.0 has been installed!
INFO tool-scons@4.40700.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (5.4.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.40407.240606 (4.4.7)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 2.35.0-20220830
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
ModuleNotFoundError: No module named 'pkg_resources':
  File "C:\Users\David\AppData\Local\Programs\Python\Python312\Lib\site-packages\platformio\builder\main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\David\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\David\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\David\.platformio\platforms\espressif32\builder\main.py", line 312:
    target_elf = env.BuildProgram()
  File "C:\Users\David\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Users\David\AppData\Local\Programs\Python\Python312\Lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\David\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Users\David\AppData\Local\Programs\Python\Python312\Lib\site-packages\platformio\builder\tools\piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\David\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Users\David\AppData\Local\Programs\Python\Python312\Lib\site-packages\platformio\builder\tools\piobuild.py", line 347:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "C:\Users\David\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 676:
    return method(*args, **kw)
  File "C:\Users\David\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\David\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\David\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 29:
    import pkg_resources
============================================= [FAILED] Took 104.77 seconds =============================================

David@DavidPC MINGW64 ~/esphome-jk-bms-can (main)

Sleeper85, 2024-08-20:
You need to install the `pkg_resources` module.

dmsims, 2024-08-20:
OK I found the esphome-jk-bms-can\config\config_atom-s3-lite-esp32-s3.yaml file and updated both the main Wire and BLE yaml files with the info from that file

It flashes both Wire and BLE OK but constantly resets:

Auto-detected Flash size: 8MB
Flash will be erased from 0x00010000 to 0x000d1fff...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00009000 to 0x0000afff...
Compressed 792976 bytes to 505797...
Wrote 792976 bytes (505797 compressed) at 0x00010000 in 5.3 seconds (effective 1186.2 kbit/s)...
Hash of data verified.
Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=8MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
Compressed 20848 bytes to 13067...
Wrote 20848 bytes (13067 compressed) at 0x00000000 in 0.3 seconds (effective 508.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 134...
Wrote 3072 bytes (134 compressed) at 0x00008000 in 0.0 seconds (effective 558.1 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x00009000 in 0.1 seconds (effective 866.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from COM4 with baud rate 115200
[23:23:05]ESP-ROM:esp32s3-20210327
[23:23:05]Build:Mar 27 2021
[23:23:05]rst:0x7 (TG0WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
[23:23:05]Saved PC:0x400454d5
[23:23:05]SPIWP:0xee
[23:23:05]mode:QIO, clock div:1
[23:23:05]load:0x3fce3808,len:0x16c4
[23:23:05]ets_loader.c 78
[23:23:06]ESP-ROM:esp32s3-20210327
[23:23:06]Build:Mar 27 2021
[23:23:06]rst:0x7 (TG0WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
[23:23:06]Saved PC:0x400454d5
[23:23:06]SPIWP:0xee
[23:23:06]mode:QIO, clock div:1
[23:23:06]load:0x3fce3808,len:0x16c4
[23:23:06]ets_loader.c 78
[23:23:07]ESP-ROM:esp32s3-20210327
[23:23:07]Build:Mar 27 2021
[23:23:07]rst:0x7 (TG0WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
[23:23:07]Saved PC:0x400454d5
[23:23:07]SPIWP:0xee
[23:23:07]mode:QIO, clock div:1
[23:23:07]load:0x3fce3808,len:0x16c4
[23:23:07]ets_loader.c 78
[23:23:09]ESP-ROM:esp32s3-20210327
[23:23:09]Build:Mar 27 2021
[23:23:09]rst:0x7 (TG0WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
[23:23:09]Saved PC:0x400454d5
[23:23:09]SPIWP:0xee
[23:23:09]mode:QIO, clock div:1
[23:23:09]load:0x3fce3808,len:0x16c4
[23:23:09]ets_loader.c 78
[23:23:10]ESP-ROM:esp32s3-20210327
[23:23:10]Build:Mar 27 2021
[23:23:10]rst:0x7 (TG0WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
[23:23:10]Saved PC:0x400454d5
[23:23:10]SPIWP:0xee
[23:23:10]mode:QIO, clock div:1
[23:23:10]load:0x3fce3808,len:0x16c4
[23:23:10]ets_loader.c 78
[23:23:11]ESP-ROM:esp32s3-20210327
[23:23:11]Build:Mar 27 2021
[23:23:11]rst:0x7 (TG0WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
[23:23:11]Saved PC:0x400454d5
[23:23:11]SPIWP:0xee
[23:23:11]mode:QIO, clock div:1
[23:23:11]load:0x3fce3808,len:0x16c4
[23:23:11]ets_loader.c 78
[23:23:13]ESP-ROM:esp32s3-20210327
[23:23:13]Build:Mar 27 2021
[23:23:13]rst:0x7 (TG0WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)

Sleeper85, 2024-08-20:
I think the problem is that you don't have this option : `board_build.flash_mode: dio`

```YAML
# Atom S3 Lite - SKU:C124
# https://docs.m5stack.com/en/core/AtomS3%20Lite
esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf

esphome:
  platformio_options:
    board_build.flash_mode: dio # Standard uses a single line for data, Dual IO uses 2 lines for data (prevent reboot loop)
```

dmsims, 2024-08-20:
Success! thank you so much

dmsims, 2024-08-21:
S3 Lite keeps going in to Safe mode (with Wire version) - log attached
[safe mode log.txt](https://github.com/user-attachments/files/16689605/safe.mode.log.txt)

dmsims, 2024-08-25:
Ok thought I would update this:

I Flashed this many times and eventually it worked

Symptoms of it not working:

Remote logs fail
OTA fails
Integration into HA fails or works and show no entities
but
Ping works
HA discovery works

Sleeper85, 2024-08-25:
This looks like @ChrisG's problem but you flashed the old version only one BMS 1.17.5 right?

dmsims, 2024-08-25:
Yes 1 BMS 1.17.5

Just trying YamBMS 1.4.3 and HA integration does not work :(

Sleeper85, 2024-08-25:
But you confirm having had the same problem with version a bms 1.17.5?

Can you test YamBMS 1.4.1 (see my post on the forum).

dmsims, 2024-08-25:
Yes confirm same issue with 1.17.5

luckylinux, 2024-08-26:
@dmsims: looking at your Log there is something blocking the `LoopTask` and the Watchdog is triggered.

I had an "edge case" (I described it in the [Forum](https://diysolarforum.com/threads/yambms-jk-bms-can-with-new-cut-off-charging-logic-open-source.79325/page-75#post-1188712)) where the only Thing left to do was to increase the Watchdog Timeout (I increased mine to 30s which is somewhat risky):
```
 esp32:
  board: {{board}}
  variant: {{variant}}
  framework:
    type: esp-idf
    
    # Introduced on 2024-05-17 while trying to solve a Watchdog Triggering on WiFi Connection Loss Issue
    sdkconfig_options:
      CONFIG_ESP_TASK_WDT: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "30"
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
```

You might want a Value < 30s for your use Case though ...

You also seem to have a weird WiFi Authentication Failed Issue though, are you sure you don't have some "Prefer 5G / Band Steering", WPA3/WPA2 Transition (or WPA3 only), etc Setting/Features enabled in your Router/AP? 

```
[10:15:09][W][wifi_esp32:670]: Event: Disconnected ssid='thesims' bssid=E4:FA:C4:B3:73:23 reason='Authentication Failed'
```

Sleeper85, 2024-08-26:
@luckylinux 

Not connecting a canbus board has the effect of slowing down the execution of the code when sending frames, it is possible that this comes from there.

luckylinux, 2024-08-26:
@Sleeper85: I see you are setting the default Protocol in https://github.com/Sleeper85/esphome-jk-bms-can/blob/development/packages/yambms/yambms_canbus.yaml#L72  to Pylon + (in the Forum I see you just changed to "Disabled") ... 

Do you mean that the ESP32 is in some state of Queueing up the CAN Frames but, without "Driver" Board (CAN Adapter) Connected, it basically fills up the Queue and/or simply it causes the CAN messages to "Time Out" ?

That sounds kinda similar to what I had using an USB CANbus Adapter (CAN Frames queueing up indefinitively and `ifconfig` swhowing them as Transmitted/Received), but the CAN Adapter (which was installed on an AC-Powered USB Hub) didn't do anything. Only solution was to remove the AC-Power Cable from the USB Hub and reboot the Raspberry Pi.

Unrelated I know, but I guess it's generic CAN bus "Issue" (or Feature).

Sleeper85, 2024-08-26:
@luckylinux 

Selecting `Disabled` protocol does not send any frames.

Sleeper85, 2024-08-26:
https://github.com/esphome/issues/issues/3317

luckylinux, 2024-08-26:
> @luckylinux
> 
> Selecting `Disabled` protocol does not send any frames.

Yes but you just changed that now. Before it defaulted to Pylon+, didn't it ?

Sleeper85, 2024-08-26:
> > @luckylinux
> > Selecting `Disabled` protocol does not send any frames.
> 
> Yes but you just changed that now. Before it defaulted to Pylon+, didn't it ?

Yes just now for testing...

luckylinux, 2024-08-26:
If you can link me to the 1.4.3 Version the other People tested ... I'll be giving it a quick Try on the Atom S3 Lite.

I'll use latest from "development" Branch otherwise.

Sleeper85, 2024-08-26:
The version tested and causing problems is the one from the development branch and probably all the old versions too because @dmsims  has the same problem with 1.17.5.

But we should not exclude a different behavior following all the latest updates of esphome.

luckylinux, 2024-08-26:
I'm currently on ESPHome `2024.7.3` but `2024.8.0` is also just out !

luckylinux, 2024-08-26:
@Sleeper85 

Uhm not sure what is going on ... Probably my Routing (I am doing a test with a Test AP here in the House while Home Assistant in on a Subnet 3 Hops away in the Garage).

And of course the more Local Homeassistant Server is full and has been shut down :disappointed: .

```
2024-08-26 23:36:26.789 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised while updating state of sensor.yambms_yambms_jk_bms_1_temperature_sensor_1, topic: 'yambms/sensor/yambms_jk-bms_1_temperature_sensor_1/state' with payload: b'nan'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 383, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 675, in state
    raise ValueError(
ValueError: Sensor sensor.yambms_yambms_jk_bms_1_temperature_sensor_1 has device class 'temperature', state class 'measurement' unit '°C' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-finite value: 'nan'
2024-08-26 23:36:26.805 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised while updating state of sensor.yambms_yambms_jk_bms_1_temperature_sensor_2, topic: 'yambms/sensor/yambms_jk-bms_1_temperature_sensor_2/state' with payload: b'nan'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 383, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 675, in state
    raise ValueError(
ValueError: Sensor sensor.yambms_yambms_jk_bms_1_temperature_sensor_2 has device class 'temperature', state class 'measurement' unit '°C' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-finite value: 'nan'
2024-08-26 23:36:26.814 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised while updating state of sensor.yambms_yambms_jk_bms_1_min_cell_voltage, topic: 'yambms/sensor/yambms_jk-bms_1_min_cell_voltage/state' with payload: b'nan'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 383, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 675, in state
    raise ValueError(
ValueError: Sensor sensor.yambms_yambms_jk_bms_1_min_cell_voltage has device class 'voltage', state class 'measurement' unit 'V' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-finite value: 'nan'
2024-08-26 23:36:26.816 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised while updating state of sensor.yambms_yambms_jk_bms_1_max_cell_voltage, topic: 'yambms/sensor/yambms_jk-bms_1_max_cell_voltage/state' with payload: b'nan'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 383, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
The Device is claimed to be added to Home Assistant but it isn't Visible anywhere in "ESPHome" -> "Devices".

So I cannot add it and cannot delete it from the GUI I guess. Already tried to restart Home Assistant ...

[20240826_esphome_logs_from_METEO-STATION_to_ROCK-5B.log](https://github.com/user-attachments/files/16754179/20240826_esphome_logs_from_METEO-STATION_to_ROCK-5B.log)

Let me see if I can manage to get the local Development Server back up. Otherwise I'll do a quick test in the Garage tomorrow ...

luckylinux, 2024-08-26:
@Sleeper85: I managed to give it another go on the other Local Home Assistant Server.

Device is successfully added and listed, BUT no Entities available.

[image]

Slighly different Logs I guess ...
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun home-assistant (no readiness notification)
s6-rc: info: service legacy-services successfully started
2024-08-27 00:20:47.892 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration ltss which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-27 00:20:47.894 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration danfoss_ally which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-27 00:20:47.895 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration eloverblik which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-27 00:20:50.929 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'eloverblik' calls async_forward_entry_setup for integration, eloverblik with title: Eloverblik 571313113160422714 and entry_id: 75962ca868d4ae901845c6b75104c09c, which is deprecated and will stop working in Home Assistant 2025.6, await async_forward_entry_setups instead at custom_components/eloverblik/__init__.py, line 40: hass.async_create_task(, please report it to the author of the 'eloverblik' custom integration
2024-08-27 00:20:53.513 WARNING (MainThread) [homeassistant.helpers.frame] Detected code that calls async_forward_entry_setup for integration eloverblik with title: Eloverblik 571313113160422714 and entry_id: 75962ca868d4ae901845c6b75104c09c, during setup without awaiting async_forward_entry_setup, which can cause the setup lock to be released before the setup is done. This will stop working in Home Assistant 2025.1. Please report this issue.
Stack (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 674, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 641, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1990, in _run_once
    handle._run()
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2161, in async_forward_entry_setup
    _report_non_awaited_platform_forwards(entry, "async_forward_entry_setup")
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1201, in _report_non_awaited_platform_forwards
    report(
  File "/usr/src/homeassistant/homeassistant/helpers/frame.py", line 151, in report
    _LOGGER.warning(msg, stack_info=True)
2024-08-27 00:20:53.564 WARNING (Recorder) [homeassistant.components.recorder.migration] The database is about to upgrade from schema version 43 to 45
2024-08-27 00:20:53.697 WARNING (Recorder) [homeassistant.components.recorder.migration] Modifying columns data_id, event_type_id in table events. Note: this can take several minutes on large databases and slow machines. Please be patient!
2024-08-27 00:20:54.031 WARNING (Recorder) [homeassistant.components.recorder.migration] Modifying columns event_id, old_state_id, attributes_id, metadata_id in table states. Note: this can take several minutes on large databases and slow machines. Please be patient!
2024-08-27 00:20:54.042 ERROR (SyncWorker_0) [aiodhcpwatcher] Cannot watch for dhcp packets: [Errno 1] Operation not permitted
2024-08-27 00:20:54.111 ERROR (MainThread) [homeassistant.components.button] Platform mqtt does not generate unique IDs. ID ESPbuttonjk-bms_esp32_restart already exists - ignoring button.jk_bms_bat02_jk_bms_esp32_restart
2024-08-27 00:20:54.819 ERROR (MainThread) [homeassistant.components.select] Platform mqtt does not generate unique IDs. ID ESPselectcan_bms_name already exists - ignoring select.jk_bms_bat02_can_bms_name
2024-08-27 00:20:54.820 ERROR (MainThread) [homeassistant.components.select] Platform mqtt does not generate unique IDs. ID ESPselectcan_protocol already exists - ignoring select.jk_bms_bat02_can_protocol
2024-08-27 00:20:55.407 WARNING (MainThread) [homeassistant.components.mqtt.sensor] Invalid state message '1723564032.0' from 'jk-bms-bat02/sensor/jk-bms-bat02_last_complete_charge/state'
[... cut ...]
2024-08-27 00:21:14.992 WARNING (MainThread) [homeassistant.components.mqtt.sensor] Invalid state message '1723564032.0' from 'jk-bms-bat02/sensor/jk-bms-bat02_last_complete_charge/state'

```

[20240826_esphome_logs_from_METEO-STATION_to_Local_Podman_Container_Same_Subnet.log](https://github.com/user-attachments/files/16754398/20240826_esphome_logs_from_METEO-STATION_to_Local_Podman_Container_Same_Subnet.log)

(Same Configuration)

Not sure if this Helps you or ...

At least it's NOT triggering in my Case (but I'm also NEITHER reading the BMS NOR talking to the Inverter).

The Entities don't show Up in Home Assistant ...

Sleeper85, 2024-08-27:
I will do another test tonight by adding a can board and then a second one. I have a SN65HVD230 and a MCP2515 on my PVbrain2 PCB.

luckylinux, 2024-08-27:
It's just to see if it's a CAN Controller Issue (CAN Controller of the ESP32[-S3]) or if it also happens over SPI. But like ChrisG said on the Forum, apparently also the MCP2515 causes the Issue.

I would have thought that with the MCP2515 the "onboard" ESP32[-S3] CAN/TWAI Controller would NOT be used since then the communication would only happen over SPI with the MCP2515.

But maybe I'm wrong ...

Sleeper85, 2024-09-02:
CANBUS 2.3.4 : Fixed bug of canbus link validation without inverter connected

closed 2024-09-02 by Sleeper85