Sleeper85/esphome-jk-bms-can #72: Question about SPI0, SPI1, SPI2 and SPI3 of ESP32-S3

txubelaxu, 2024-09-06:
I have BMS JK-PBx working via CAN bus with Victron directly. But I would test your solution.

My config is multi-bms_JK-PB_RS485_mode2.yaml. And my device is this one:
[image]
It is similar to MCP2515.

I have seen this code:
[image]

And this other info:
[image]

First question: Are the pin numbers OK?
Second question: About the "spi1_bus" name, is it correct or is "spi3_bus" name correct?
Third question: Do I need to enable something in my main yaml?

Many Thanks!!

Sleeper85, 2024-09-06:
The code example you found is the one used with PVbrain2.

The name id `spi1_bus` is not important, it is just an id, you can call it whatever you want.
The most important is not to use the GPIOs assigned for Flash and PSRAM.

Then, you have to declare a can bus with this id `canbus_node1` because it is the one used by default with YamBMS.

[I just added an example board for MCP2515.](https://github.com/Sleeper85/esphome-jk-bms-can/blob/development/packages/board/board_esp32-s3-devkitc-1_mcp2515.yaml)

```YAML
spi:
  # BUS SPI 1
  - id: bus_spi1
    mosi_pin: 11
    miso_pin: 12
    clk_pin: 13

canbus:
  # CANBUS NODE 1
  - platform: mcp2515
    spi_id: bus_spi1
    id: canbus_node1
    cs_pin: 21
    can_id: 2
    bit_rate: 500kbps
```

In the main YAML you need to import the correct board YAML.

```YAML
# +--------------------------------------+
# | Packages                             |
# +--------------------------------------+
packages:
  device_board: !include packages/board/board_esp32-s3-devkitc-1_mcp2515.yaml
  device_base: !include packages/base/device_base.yaml
```

Sleeper85, 2024-09-06:
On your CAN board there are two CAN outputs.

If necessary you can send the YamBMS infos to two different inverters, even using two different CAN protocols.

You need to create 2x canbus_node and import 2x canbus.

```YAML
  canbus1: !include
    file: packages/yambms/yambms_canbus.yaml
    vars:
      # YamBMS ID
      yambms_id: 'yambms1'
      # CANBUS vars
      canbus_id: 'canbus1'
      canbus_name: 'CANBUS 1'
      canbus_node_id: 'canbus_node1'
      canbus_light_id: 'esp_light'
      # The CANBUS link will be considered down if no response from the inverter (ID 0x305) for 5s
      # The Deye inverter sends the ACK (can_id 0x305) only when it receives the can_id 0x356
      canbus_link_timer: '5'

  canbus2: !include
    file: packages/yambms/yambms_canbus.yaml
    vars:
      # YamBMS ID
      yambms_id: 'yambms1'
      # CANBUS vars
      canbus_id: 'canbus2'
      canbus_name: 'CANBUS 2'
      canbus_node_id: 'canbus_node2'
      canbus_light_id: 'esp_light'
      # The CANBUS link will be considered down if no response from the inverter (ID 0x305) for 5s
      # The Deye inverter sends the ACK (can_id 0x305) only when it receives the can_id 0x356
      canbus_link_timer: '5'
```

txubelaxu, 2024-09-06:
Many thanks, @Sleeper85. I have been making a lot of tests after your answers, but it is not working for me for now.
I have connected the CAN cable to the Victron Cerbo GX, and to the CAN transceiver H/L/G.
I have checked the wires and are OK: positions and electrically.
I have tested with and without "CAN loads".

The main yaml goes OK at compiling and running times.

```[23:48:53][I][main:537]: Charge Status : Float
[23:48:53][I][canbus:545]: send can_id: 0x35F hex: 0 0 2 2 60 2 0 0
[23:48:53][I][canbus:625]: send can_id: 0x372 hex: 2 0 0 0 0 0 0 0
[23:48:53][I][canbus:656]: send can_id: 0x373 hex: 24 d 92 d 2f 1 2f 1
[23:48:53][I][canbus:673]: send can_id: 0x374 [ASCII] Min cell voltage ID : 212
[23:48:53][I][canbus:674]: send can_id: 0x374 hex: 32 31 32 0 0 0 0 0
[23:48:53][I][canbus:691]: send can_id: 0x375 [ASCII] Max cell voltage ID : 210
[23:48:53][I][canbus:692]: send can_id: 0x375 hex: 32 31 30 0 0 0 0 0
[23:48:54][I][canbus:705]: send can_id: 0x376 [ASCII] Min Temp. Sensor ID : 2
[23:48:54][I][canbus:723]: send can_id: 0x377 [ASCII] Max Temp. Sensor ID : 1
[23:48:54][I][canbus:743]: send can_id: 0x379 hex: 60 2 0 0 0 0 0 0
[23:48:54][I][canbus:754]: send can_id: 0x382 [ASCII] Product : YamBMS
[23:48:54][I][canbus:264]: send can_id: 0x35E ASCII : BYD
[23:48:54][I][main:537]: Charge Status : Float
[23:48:54][I][canbus:296]: send can_id: 0x351 hex: 26 2 b0 4 a0 5 a8 1
[23:48:54][I][canbus:336]: send can_id: 0x355 hex: 62 0 64 0 48 26 bf 17
[23:48:54][I][canbus:375]: send can_id: 0x356 hex: a9 15 0 0 2f 1 0 0
[23:48:54][I][canbus:522]: send can_id: 0x35A hex: 0 0 0 0 0 0 0 0
[23:48:54][I][canbus:545]: send can_id: 0x35F hex: 0 0 2 2 60 2 0 0
[23:48:55][I][canbus:625]: send can_id: 0x372 hex: 2 0 0 0 0 0 0 0
[23:48:55][I][canbus:656]: send can_id: 0x373 hex: 25 d 92 d 2f 1 2f 1
[23:48:55][I][canbus:673]: send can_id: 0x374 [ASCII] Min cell voltage ID : 212
[23:48:55][I][canbus:674]: send can_id: 0x374 hex: 32 31 32 0 0 0 0 0
[23:48:55][I][canbus:691]: send can_id: 0x375 [ASCII] Max cell voltage ID : 210
[23:48:55][I][canbus:692]: send can_id: 0x375 hex: 32 31 30 0 0 0 0 0
[23:48:55][I][canbus:705]: send can_id: 0x376 [ASCII] Min Temp. Sensor ID : 2
[23:48:55][I][main:537]: Charge Status : Float
```

I think that the info is not been reading by the Cerbo GX.

I am doing more tests.

Sleeper85, 2024-09-07:
I know two guys using it with Victron.

Are you using the latest version 1.4.4.B?

This version fixes several bugs.

txubelaxu, 2024-09-07:
I have updated the "development" branch code. I have compiled and run, but nothing...

main yaml:
```
  canbus1: !include
    file: packages/yambms/yambms_canbus.yaml
    vars:
      # YamBMS ID
      yambms_id: 'yambms1'
      # CANBUS vars
      canbus_id: 'canbus1'
      canbus_name: 'CANBUS 1'
      canbus_node_id: 'canbus_node1'
      canbus_light_id: 'esp_light'
      # The CANBUS link will be considered down if no response from the inverter (ID 0x305) for 5s
      # The Deye inverter sends the ACK (can_id 0x305) only when it receives the can_id 0x356
      canbus_link_timer: '5'
```

board yaml:
```
spi:
  - id: bus_spi1
    mosi_pin: GPIO11
    miso_pin: GPIO13
    clk_pin:  GPIO12    

canbus: 
  - platform: mcp2515
    spi_id: bus_spi1
    id: canbus_node1
    cs_pin: GPIO10
    can_id: 2
    bit_rate: 500kbps

#canbus:
# # # CANBUS NODE 1
#  - platform: esp32_can
#    id: canbus_node1
#    tx_pin: 38
#    rx_pin: 39
#    can_id: 1
#    bit_rate: 500kbps
```

I have set "Victron" at protocol dropbox:
[image]

I have tested with 500kbps and 250kbps; can_id:1 and can_id:2; and Cerbo GX distinct protocols... but I get only "errors" in the CANBUS.

[image]

txubelaxu, 2024-09-07:
Could those guys you know share their settings with me, please, so I can test the whole config RS485 --> CAN --> Victron Cerbo GX.

Many thanks in advance!!

Sleeper85, 2024-09-07:
[You will find them in the list of supported inverters.](https://github.com/Sleeper85/esphome-jk-bms-can/blob/development/documents/README/Supported_devices.md#supported-inverter)

@dmsims is here on Github.

[image]

Is the Cerbo GX compatible with SN65HVD230 CAN transceiver ?

[image]

dmsims, 2024-09-07:
@txubelaxu

AS a test can you try BLE ?

If you have 1 Can cable plugged into the Cerbo you must use a terminator

txubelaxu, 2024-09-07:
Thanks @Sleeper85

> Is the Cerbo GX compatible with SN65HVD230 CAN transceiver ?

I think it is compatible. Andy is using it ("Today we replace the 'normal' CAN adapter on the Raspberry Pi running my Victron installation with the double port insulated CAN Hat.") Here the video for more info: https://www.youtube.com/watch?v=bGmB97ceiKg

The micros are exactly the same:
[image]

Maybe, Am I missing something about the unused "interruption pin"? The Raspberry is using it.

txubelaxu, 2024-09-07:
Hi @dmsims, many thanks for your answer. I see that you are using a Cerbo GX as well.

> AS a test can you try BLE ?

Well, I am using RS485 and that part is working perfect. I could try with BLE but I do not think it make canbus work.

> If you have 1 Can cable plugged into the Cerbo you must use a terminator

Yes, I have tested with and without network terminator. But nothing...
What speed are you using? 250kbps or 500kbps?
What protocol have you set in the Cerbo GX?
[image]

Could you share the board code that you are using (only "spi" and "canbus" fragments):
```
spi:
  - id: bus_spi1
    mosi_pin: GPIO11
    miso_pin: GPIO13
    clk_pin:  GPIO12    

canbus: 
  - platform: mcp2515
    spi_id: bus_spi1
    id: canbus_node1
    cs_pin: GPIO10
    can_id: 2
    bit_rate: 500kbps

#canbus:
# # # CANBUS NODE 1
#  - platform: esp32_can
#    id: canbus_node1
#    tx_pin: 38
#    rx_pin: 39
#    can_id: 1
#    bit_rate: 500kbps
``` 

Many thanks in advance!!!

Sleeper85, 2024-09-08:
@txubelaxu If you look carefully at the list of supported inverters that I sent you, you can see that @dmsims uses an `Atom S3 Lite` with `Atomic CAN base` so no SPI bus.

Concerning the Victron protocol used I let @dmsims answer you.

The other guy is using a 100% M5Stack solution (Atom S3 + Atomic CAN base + RS485 unit) with a 500kbps CAN bus.

Personally I use a PVbrain2 PCB with a MCP2515 and a SN65HVD230 on the same bus. The spi bus is configured as in the example I showed you.

txubelaxu, 2024-09-08:
I have got working now.
CAN-bus BMS LV (500 kbit/s)
I do not know why now is working and not before.

Now, some reboots detected on ESP32-S3. I have to check it.

Many thanks!!!

Patrick197030100, 2024-09-08:
hello vous avais un victron aussi surtout pas connecter en VECAN  CAR SINON VOUS BLOQUE LE SYSTÈME VICTRON !!!! et redémarrer  esp32 sinon il ne  reconnaitra  pas le bms

Xenon1970, 2024-09-08:
[image]

Xenon1970, 2024-09-08:
[image]

Xenon1970, 2024-09-08:
[image]

txubelaxu, 2024-09-08:
Well... I have been debugging and I have found how to solve the problem of restarts with last "development" version.
I have tested with 20240812 version and it works without restarts for me.

Xenon1970, 2024-09-08:
ok sorry I don't speak English it's just sleeper 85 told me that you can't get it to work you succeeded great a+

txubelaxu, 2024-09-08:
Many many thanks, @Xenon1970 !!!
Yes, I got it working.

Sleeper85, 2024-09-08:
> Now, some reboots detected on ESP32-S3. I have to check it.

@txubelaxu

What kind of problems have you seen ?
Another person on the forum reports a crash of the ESP32 using your RS485 component.

txubelaxu, 2024-09-08:
Random reboots. I do know why...
I have tested the last version of esphome-jk-bms and no reboots, but with last esphome-jk-bms-can last version I detect reboots.

If I run a previous version of esphome-jk-bms-can with same version of esphome-jk-bms, no reboots.

Sleeper85, 2024-09-09:
> Random reboots. I do know why... I have tested the last version of esphome-jk-bms and no reboots, but with last esphome-jk-bms-can version, then I detect reboots.
> 
> If I run a previous version of esphome-jk-bms-can with same version of esphome-jk-bms, no reboots.

Please be a little more specific about what you have tested and found.

Several people use YamBMS 1.4.4.B (latest version) including myself.
Whether with BLE / UART or RS485 on ESP32 or ESP32-S3 (Atom S3) we have uptimes of more than 5 days.

[image]

txubelaxu, 2024-09-09:
I have been checking (DEBUG and VERBOSE) and I have not seen the cause of reboots.
I have just upload to my development_txubelaxu_last the code that works for me without rebooting.